schreibfaul1/ESP32-MiniWebRadio #82: wishes and clarifications

valera39, 2022-04-06:
Can I have a small wish for a future update? :)
- when pressing the digital buttons of the remote control, the pressed numbers were displayed on the display.
- on the initial screen, when pressing the "OK" button, for example, the clock was displayed, pressing again switched back to the main screen
In the updated version in the audio player on the screen, the "next track" button has been changed to "play"
Special thanks for FTP.
Have a good day!

romulus1st, 2022-04-08:
These changes solve the problem of waking up from sleep mode. Waking up is done by pressing the IR OK key.
```
void ir_res(uint32_t res){
    _f_irNumberSeen = false;
    if(_state != RADIO) return;
    if(_f_sleeping == true) return;
    if(res != 0){
       setStation(res);
    }
    else{
        setStation(_cur_station); // valid between 1 ... 999
    }
    return;
}
```

```
void ir_key(const char* key){

    //if(_f_sleeping) {_f_sleeping = false;  changeState(RADIO);} // awake
    if(_f_sleeping == true && key[0] != 'k') return;
    if(_f_sleeping == true && key[0] == 'k'){ //awake
        _f_sleeping = false;
        SerialPrintfln("awake");
        setTFTbrightness(pref.getUShort("brightness"));
        changeState(RADIO);
        connecttohost(_lastconnectedhost);
        showLogoAndStationName();
        showFooter();
        showHeadlineItem(RADIO);
        showHeadlineVolume(_cur_volume);
        return;
    }
```

valera39, 2022-04-08:
Hello romulus1st!
I applied your changes, it really works.
Thank you!

schreibfaul1, 2022-04-08:
Hello everyone,
I added the improvements. The digits from the IR transmitter are now also shown on the display. I had created my own character set for this, but unfortunately never activated it.
[image]

valera39, 2022-04-08:
Hello!
I downloaded the updates but the numbers are not showing up :(

valera39, 2022-04-08:
Hello everyone, hello Wolle!
The numbers on the screen work, thanks. What about my other suggestions?
And a couple more questions: what is the AUDIOTASK_CORE parameter in common.h responsible for, for sound processing on a separate core?
Is it worth changing it?
In platformio.ini when installing on ESP32 Wrower to enable PSRAM, do I need to uncomment the parameters
DBOARD_HAS_PSRAM and mfix-esp32-psram-cache-issue
And if it were possible to use ESP-12K with 8Mb PSRAM, then it would be great, there is never too much memory :)

closed 2022-05-08 by valera39