schreibfaul1/ESP32-MiniWebRadio #83: Suggestions for improvement

valera39, 2022-04-09:
Hello everyone, hello Wolle!
Let me repeat myself a little. Is it possible to implement such algorithms:
- when you press the "OK" button, the initial screen switches to the clock, pressing again switches back to the initial screen.
If you switch the station using the buttons on the remote control in the clock mode, display the initial screen for a while, then switch again
on the screen with hours.
- loop the switching of stations using the "next" and "next" buttons on the screen and on the remote control, that is, the last station in the list, then the first
- to prevent clicks in the speakers when the radio is turned on and to reduce the noise of the amplifier in the absence of a signal
add the "mute" mode to the free GPIO, the signal from which appeared at the time of the DAC operation.
I think the radio will become more functional from these functions.
And a couple more questions: what is the AUDIOTASK_CORE parameter in common.h responsible for, for sound processing on a separate core?
Is it worth changing it?
In platformio.ini when installing on ESP32 Wrower to enable PSRAM, do I need to uncomment the parameters
DBOARD_HAS_PSRAM and mfix-esp32-psram-cache-issue
If it were possible to use ESP-12K with 8Mb PSRAM, it would be great, because there is never too much memory :)

schreibfaul1, 2022-04-09:
Hello @valera39,
the first two points are done. Now the remote control can be used properly.
Point three: With 'mute', the volume is faded in or out in stages. In addition, if AMP_ENABLED is set in common.h, the assigned GPIO is controlled. AMP_ENABLED was already present to control the amplifiers on the audio boards (e.g. AI Thinker A1S board or OLIMEX ADF) and to save energy.
The ESP32 has two cores and one can be selected with AUDIOTASK_CORE. By default, Arduino setup() and loop() runs on core 1 and therefore the display also runs on core1. I think for the Aufiotask core 0 is better.
PSRAM is not strictly required. There is one exception: FLAC with an external DAC requires PSRAM, the SRAM does not have enough capacity. Older ESP32 (these have a design flaw) require the mfix-esp32-psram-cache-issue.
ESP-12K is certainly an ESP32-S2 and would then not work because of only one core.
regards
Wolle

valera39, 2022-04-09:
Hello Wolle!
Thank you very much for the implementation of my wishes and detailed answers to my questions!
I'm using an ESP32 worower with PSRAM enabled, but I'm experiencing a brief, literally split-second, audio stutter at the moment
change the name of the song on the screen. There are also slight stutters when using the buttons on the screen. Files like .flac do not play correctly.
Maybe you can separate the sound processing and display control into different processor cores to evenly distribute the load?
Or slightly increase the size of the signal's audio buffer.
Can I make sequential playback of the files located there in the audiofiles folder? Ideally, I'd like more
the functionality of the audio player, but I still think the main thing is the radio, since the processor will not pull much.
Thanks for your work!

closed 2022-05-08 by valera39