schreibfaul1/ESP32-MiniWebRadio #84: Station Liste

newbee90, 2022-04-09:


newbee90, 2022-04-09:
Ich habe die stations.csv bearbeitet, zb Sender verschoben oder gelöscht, die Stationen werden aber im Radio nicht aktualisiert.
Der Name passt nicht zur URL, usw. Bei der alten Version habe ich das geschaft

schreibfaul1, 2022-04-09:
Die stations.csv ist eine mit TAB als Trennzeichen formatierte Liste. Mit Excel lässt sich die Tabelle am einfachsten bearbeiten. Lade die Liste auf der Webseite mit dem Button <save xlsx> herunter, editiere in Excel (oder Staroffice, Libreoffice...) und schreibe die Datei mit <load xlsx> zurück. Wenn die Tabelle ordentlich in der Webseite angezeigt wird speichere diese mit <save> auf die SD. 
vG
Wolle

newbee90, 2022-04-10:
es wird richtig angezeigt
https://share-your-photo.com/818882df97/album

closed 2022-04-11 by newbee90