schreibfaul1/ESP32-MiniWebRadio #85: loss of sound

valera39, 2022-04-10:
Hello Wolle!
Thank you very much for the implementation of my wishes and detailed answers to my questions!
I am using an ESP32 worower with -DBOARD_HAS_PSRAM enabled in platformio.ini, DAC VS1053 and ILI9488 and I have a short, literally half a second, loss of sound at the moment the information about the song name on the screen changes. This is very noticeable to the ear. Can anything be done about it?
Maybe it's possible to separate audio processing and display control into different processor cores to evenly distribute the load or increase the size of the audio buffer?
Files like .flac also play incorrectly.
Is it possible to make sequential playback of files located in the audiofiles folder? Ideally I would like more
the functionality of the audio player, but I still think the main thing is the radio, since the processor will not pull much.
Thank you for your wonderful work!

valera39, 2022-04-11:
Hello Wolle!
Your updates at the moment, will help solve the problem that I described above?
Also check the station [https://www.5-tv.ru/radio.mp3](url) does not play on both versions of the radio.
Thank you!

OK1FWG, 2022-04-11:
The latest code is causing no audio at all and regular booting of the device. Loaded latest code from you in PIO, compiled, but no avail.

schreibfaul1, 2022-04-12:
Hello everyone,
several settings led to the error:
1) in platformio I had set 'DCORE_DEBUG_LEVEL=1' (print errors only) and in vs1053ext.cpp there was a too long log_e() string which led to a memory overflow
2) the stack was too small for the audio task, which caused the crash with ssl connections.

That is now eliminated.
https://www.5-tv.ru/radio.mp3 is not a 'normal' stream, there is no audio header, but an HTML document with embedded javascript. I believe there will be a special APP for this radio station.

valera39, 2022-04-12:
Hello Wolle!
The momentary loss of sound at the moment of changing the information about the song title on the screen is still present :(

valera39, 2022-04-13:
Hello Wolle!
Updated to the latest changes at the moment Everything works fine. The sound does not disappear. Excellent!
Let me correct in main.cpp
1448: _pressBtn[2] = "/btn/Button_Right_Yellow.jpg"; _releaseBtn[2] = "/btn/Button_Right_Blue.jpg";
  on _pressBtn[2] = "/btn/Button_Last_Yellow.jpg"; _releaseBtn[2] = "/btn/Button_Last_Blue.jpg";
  I think that would be more accurate.
  It remains to add sequential playback of audio files on the SD Card and everything will be perfect, as it seems to me.
  Thank you!

closed 2022-05-08 by valera39