schreibfaul1/ESP32-MiniWebRadio #89: time sound every hour

valera39, 2022-04-30:
Hello Wolle!
 If I understand correctly, in main.cpp the line pref.putBool("pref.putBool("timeAnnouncing", 1); // Time announcement every full hour); // Time announcement every full hour
 is responsible for playing the time every hour. I set the value timeAnnouncing", 0, but time still play.
 And I will ask again, is it possible to assign PIN 25, 26 or 27 for amp_enalBand?
 Thank you!

schreibfaul1, 2022-05-06:
Hello valera39,
the hourly time announcement can now be controlled via the website.
Pins 25, 26, 27 carry the I2S protocol for an external DAC. I don't know exactly what an amp_enalBand is.

valera39, 2022-05-06:
I meant it
В common.h line 73 #define AMP_ENABLED   -1

schreibfaul1, 2022-05-07:
This is a connection that is found on audio boards. That's where the DAC and amplifiers are on the same board along with the ESP32.
You can see examples of AI Thinker's A1S board here
https://github.com/schreibfaul1/ESP32-MiniWebRadio/blob/master/additional_info/AI%20Thinker%20A1S%20Board.pdf
https://forum.espuino.de/t/esp32-audio-kit-esp32-a1s/106

Olimex ADF
https://raw.githubusercontent.com/schreibfaul1/ESP32-audioI2S/master/examples/ESP32-ES8388/ESP32-ADF_Rev_C.svg

TTGO TAudio
https://github.com/schreibfaul1/ESP32-audioI2S/tree/master/examples/TTGO-TAudio

valera39, 2022-05-07:
Thanks for the detailed answer!
I just wanted to use the free GPIO output of the ESP32 to control the MUTE mode of an external amplifier.
I have some ideas for the design and organization of the OSD. If you are interested, I am ready to describe them.
Have a nice day!

closed 2022-05-11 by valera39