schreibfaul1/ESP32-MiniWebRadio #90: no image on the display

valera39, 2022-05-04:
Hello Wolle!
My config:version 2.2o, ESP32wroom,VS1053,ILI9488.
When the power is turned on, the radio boots up, works and controls, but the screen is dark. The display starts to show
information only after pressing the RST button on the ESP32. This was not observed in previous versions.
Please also reply to my previous post.
Laut.fm stations work great. Thanks for support!

siffup, 2022-05-04:
Try to connect TFT-BL directly to VCC, this should solve your problem.

valera39, 2022-05-04:
Of course, this is the simplest solution, but it does not correct the problem, since other functions of the radio are lost. I will wait for corrections from the author. It seems to me that it is the order of loading the elements of the radio, because in version 2.2m everything was fine. I sewed again at 2.2m and everything works. But in version 2.2o, the problem of the SSL tie is solved.
I really look forward to corrections.

valera39, 2022-05-05:
Hello Wolle!
Your update solved my problem completely. There are only questions left in the previous post.
Thank you!

schreibfaul1, 2022-05-05:
Hello valera39,
I made a bug myself, the audio file was never closed in the vs1053 library. I replaced some XMLHttpRequests with Websocket requests, which are more reliable.
The backlight pin is PWM controlled and cannot drive the display's LEDs directly. Most displays have a transistor to control the BL for this.

valera39, 2022-05-05:
Okay, and my issues are https://github.com/schreibfaul1/ESP32-MiniWebRadio/issues/89

closed 2022-05-08 by valera39