schreibfaul1/ESP32-MiniWebRadio #91: Image look  blurry, colors appear washed out

siffup, 2022-05-08:
Hey Wolle,

still got the problem with the images. Using a 4 inch waveshare with ili9486 that works like a charm with the elder code (V1). Can you pleas try to fix the problem?

Thank you and keep it up, your doing a great job!

[image]

valera39, 2022-05-08:
Hello, I'll try to advise.
In file common.h try changing the value #define DISPLAY_INVERSION

schreibfaul1, 2022-05-08:
Hello siffup,
the tip from valera39 could help, alternate
I think we've had the gamma curve discussion here before. Please look in lib/tftlib/src/src/tft.cpp from line 260 there is an alternative gamma curve. I would be interested to know if this solves the problem.

siffup, 2022-05-08:
@ valera39: that`s what I tried first, it didn`t work. But thx for the advice.
@ Wolle. Bingo!!! Changed both gamma control to the alternativ parameters and the freq to 20000000 and it works.
Thank you!!!
[image]

siffup, 2022-05-09:
Problem solved.

closed 2022-05-09 by siffup