schreibfaul1/ESP32-MiniWebRadio #92: MP3 player stops after first song

siffup, 2022-05-10:
I uploaded a couple of sogs to the MMC, but the player stops after the first song.
Does line 1721 to 1729 usually starts the next song, as it is seems to be a duplicate...
void vs1053_eof_mp3(const char *info){                  // end of mp3 file (filename)
    _f_eof = true;
    if(startsWith(info, "alarm")) _f_eof_alarm = true;
    SerialPrintfln("end of file: %s", info);
}
void audio_eof_mp3(const char *info){                  // end of mp3 file (filename)
    _f_eof = true;
    if(startsWith(info, "alarm")) _f_eof_alarm = true;
    SerialPrintfln("end of file: %s", info);
}
[image]

schreibfaul1, 2022-05-11:
thanks, I haven't planned anything like this before. In the future there will be a playlist, whereby the audio files can be on the SD card or on a server.

siffup, 2022-05-11:
ok perfect, will look forward...

siffup, 2022-05-16:
Very nice!!! m3u is working flawless. Is it possible to add a function so I can skip back and forth from sondg to song in the playlist?

siffup, 2022-06-16:
Solved by Schreibfaul, thank you

closed 2022-06-16 by siffup