schreibfaul1/ESP32-MiniWebRadio #97: Sound varies

riks5, 2022-05-23:
Hi,

I intent to build your latest version on a PCB.
At the moment testing a breadboard setup.
Hardware: ESP32Wrover with psRam + SD-card + software DAC with amplifier MAX98357, no display attached.
IDE: Platformio on Eclipse.

So far so good: up and running.

A build with or without psram: soundlevel sometimes varies.
Switching between radio-player and files-player brings normal soundlevel back.
On the webpage nor serial, I ever saw a change of values during sounlevel alterations.
Volume indicates constant level 5, but you hear soundlevel go up or go down.

What am I doing wrong?
A known issue?
Have you any ideas or tips to approche this  problem?

Thanks for your time, advise and great project!

Best regards.
Rik.

schreibfaul1, 2022-05-24:
The MAX98357 is a DAC with an amplifier and you need 2 for stereo?
Perhaps the pin that distinguishes between the right and left channels is not connected?

riks5, 2022-05-25:
Hi,

Thanks for the quick reponse and your assistance.

The dac+amp is a adafruit module 3006.
I connected it out of the box and the default config is then what they call "mono-mode"( L1/2 + R1/2).
Today I changed one module to leftchannel-mode and another module to rightchannel-mode.
In all 3 modes the variations in soundlevel are present.

I will change my pcb-design and go for another module.
Any i2s-model you would advise?

Best regards.
Rik.

EFWob, 2022-06-14:
I had similar problems with a breadboard setup. So may be its just a case of bad electric connection with dupont wires. In my case, the problem was gone after I soldered the connections between ESP and DAC...

closed 2023-04-14 by stale[bot]