schreibfaul1/ESP32-MiniWebRadio #101: esp32-s3-12k "WiFi credentials are not correct"

valera39, 2022-06-28:
Hello!
Decided to use ESP32-S3-12K
After flashing the latest version, I get this error. How to be?

schreibfaul1, 2022-07-01:
Sorry I missed it, so much is coming in here, not just via GitHub.
I can understand the error, elimination coming soon.

schreibfaul1, 2022-07-01:
Hi valera39,
on the ESP32, the symbol SPI is linked to VSPI by default. The ESP32-S3-12K uses VSPI internally. HSPI and FSPI are free. I used FSPI but didn't rename to source code everywhere. Thanks for the hint.

valera39, 2022-07-02:
Hi Wolle!
Thanks for the changes, the icon is now displayed properly. For the second part of my question, I made a video [https://dropmefiles.com/gfdBI](url) at 48 seconds you can hear
how the sound disappears when the track name is changed on the display. I set different AUDIOTASK_PRIO and AUDIOTASK_CORE values in common.h but this
did not help. In the first version of your radio, there was no such problem at all.
I hope for an answer and thank you for paying attention to my questions, good luck!

closed 2022-07-03 by valera39