schreibfaul1/ESP32-MiniWebRadio #105: Audio interruptions

valera39, 2022-07-03:
Hi Wolle!
Thanks for the changes, the icon is now displayed properly. For the second part of my question, I made a video https://dropmefiles.com/gfdBI at 48 seconds you can hear
how the sound disappears when the track name is changed on the display. I set different AUDIOTASK_PRIO and AUDIOTASK_CORE values in common.h but this
did not help. In the first version of your radio, there was no such problem at all.
I hope for an answer and thank you for paying attention to my questions, good luck!

schreibfaul1, 2022-07-03:
Hello valera39,
yes i can confirm. I can draw graphics on the display without any problems, but as soon as the stream title changes, the data stream briefly breaks off. I'll take a closer look at that in the next few days.

valera39, 2022-07-03:
Thank you, I'll wait for a solution

schreibfaul1, 2022-07-04:
The audio task wrote the stream title on the display. I decoupled that and the audio task is no longer slowed down.

closed 2022-08-03 by valera39