schreibfaul1/ESP32-MiniWebRadio #107: WiFi signal strength

valera39, 2022-07-12:
Hi Wolle!
Can I display the Wi-Fi signal strength?
Thank you!

schreibfaul1, 2022-07-15:
added rssid to the footer, is updated every second, new bitmaps are used, the  folder 'common'  must be updated for this
best regards
Wolle

valera39, 2022-07-15:
Thanks, I'll try. And my previous idea is not suitable for you to implement?

valera39, 2022-07-16:
Hi Wolle!
If I had not noticed that you changed platformio.ini, then I would not have known that you solved the problem described above.
I thought you would let me know that the problem is solved.
Thanks for reading

schreibfaul1, 2022-07-16:
Hi valera39,
I have 3 devices with different hardware and cannot understand the error described.
What configuration do you have, DAC or VS1053?
In PlatformIO, 'Clean All' often helps, deleting everything precompiled and fetching the libraries again.
If that doesn't help, please set DCORE_DEBUG_LEVEL to 3 in the ini and send me the log.

valera39, 2022-07-16:
Hi Wolle!
Sorry for not providing complete information.
I have several of your radios built, the latest on ESP32-S3, VS1053, ILI9488. Others on ESP32 WROWER, VS1053, ILI9341.
With the latest changes and now everything is working fine. If there are problems, I will definitely provide a log.
Thanks for your work!
Is the proposal for vu meter not feasible?

schreibfaul1, 2022-07-16:
very good,
whether a vu meter is feasible I can't say yet. One could determine the amplitude of the I2S signal (when using a DAC), but due to the DMA buffer there is a time delay which depends on the filling level. The VS1053 doesn't work that way, maybe there are registers to read out?

valera39, 2022-07-16:
Unfortunately, I'm not strong in programming, but I'll try to understand the scope of my capabilities.
There is a project where this has already been implemented, take a look 
[https://github.com/e2002/yoradio/](url)
[https://4pda.to/forum/index.php?showtopic=1010378&st=1880#entry112992611](url)
[ https://4pda.to/forum/dl/post/26007322/vu494.mp4](url)

KT315E, 2022-07-29:
> К сожалению, я не силен в программировании, но постараюсь понять масштаб своих возможностей. Так же у меня есть в наличии PCM5102A, если на VS1053 это реализовать невозможно Есть проект где это уже реализовано, гляньте [https://github.com/e2002/yoradio/ ](url) [https://4pda.to/forum /index.php?showtopic=1010378&st=1880#entry112992611 ](url) [https://4pda.to/forum/dl/post/26007322/vu494.mp4](url)

Привет  Валера
А по ссылкам нет ни чего(

valera39, 2022-07-29:
А если так: https://github.com/e2002/yoradio/
                  https://4pda.to/forum/index.php?showtopic=1010378&st=1880#entry112992611

closed 2022-08-03 by valera39