schreibfaul1/ESP32-MiniWebRadio #111: 2.8 TFT SPI V1.2

salvi5, 2022-08-10:
Hallo. Ich habe o.g. TFT Display mit dem ILI9341 und bekomme den Touch in keiner Weise zum laufen. Die Verkabelung ist zig-mal geprüft, alle TP-Versionen ebenfalls. Ich bin nun etwas ratlos.

schreibfaul1, 2022-08-11:
Ich habe mehrere Displays ohne Probleme im Einsatz, die Verkabelung ist:
[image]

salvi5, 2022-08-11:
Vielen Dank, der Touch funktioniert jetzt offenbar. Deinen Anschlussplan sehe ich hier zum ersten Mal. Der Touch ist allerdings spiegelverkehrt und ich habe bei meiner Suche mal wieder nichts dazu gefunden. Hast Du dazu auch einen Tip?

Dankbare Grüße
Mike

salvi5, 2022-08-11:
Erledigt. Habe es gefunden. Ganz vielen Dank für die Software und die Hilfe!!!
Gruß, Mike

Dudinu, 2022-10-16:
Hi! Everything seems to work fine, except, like Salvi5, the touchscreen is mirrored. Any ideas ? Thanks, Sorin

Dudinu, 2022-10-16:
TP_VERSION = 0 & TP_ROTATION = 1 worked for me. Thank you for your great project. Sorin

Jefke1, 2023-02-22:
Hi,
First of all thanks for sharing this project.
Everything works perfectly except the TFT ILI9341.
Startup goes well, transmitter information is shown, but after 5 sec the display turns white. after reset, screen is displayed in the wrong rotation, used another display of the same type. same problem. I did the wiring as described above. It is the new type, see link. 
Thanks in advance Jef.
https://esphome.io/components/touchscreen/xpt2046.html

closed 2024-03-12 by github-actions[bot]