schreibfaul1/ESP32-MiniWebRadio #112: Constant reboot due to station

valera39, 2022-08-14:
Hello Wolle!
When switching to this station https://sovetskie-pesni.ru:8005/stream 
The radio starts to reboot constantly. Only erasing the ESP32 and re-flashing helps. Please try it yourself.
I would also like your response to my previous [post.]https://github.com/schreibfaul1/ESP32-MiniWebRadio/issues/110(url)

schreibfaul1, 2022-08-15:
This is strange, the station works with the I2S Lib but not with VS1053. I imported some routines from the I2S Lib to keep the libraries consistent. Now the error seems to have been eliminated, I hope I didn't add other errors.
Bitrate was displayed in previous versions. This didn't work really well because the bitrate doesn't have to be constant, many sources use a variable bitrate.

valera39, 2022-08-15:
That is, on VS 1053 this station will not work?

closed 2022-08-20 by valera39