schreibfaul1/ESP32-MiniWebRadio #115: SDMMC Errors

gerthalberg, 2022-09-02:
Getting up and running on a bare A1S chip I get the following:

rst:0x1 (POWERON_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1344
load:0x40078000,len:13836
load:0x40080400,len:3608
entry 0x400805f0
ESP32 Chip: ESP32-D0WDQ5
Arduino Version: 2.0.4
ESP-IDF Version: 4.4.1
ARDUINO_LOOP_STACK_SIZE 8192 words (32 bit)

00:00:00 
00:00:00        ***************************    
00:00:00        *     MiniWebRadio V2     *    
00:00:00        ***************************    
00:00:00 
00:00:00 setup: ....  Arduino is pinned to core 1
00:00:00 tft_info: .  init ILI9488
00:00:00 setup: ....  Init SD card
00:00:00 setup: ....  SD card found
00:00:01 setup: ....  seek for stations.csv
00:00:01 setup: ....  stations.csv found
00:00:01 setup: ....  seek for WiFi networks
00:00:04 WiFI_info:   Connecting WiFi...
00:00:04 setup: ....  connected to WiFimodem-ABBE, IP address is 192.168.0.51
00:00:04 ftpServer:   Buffers allocated: 4622 bytes
00:00:04 rtime_info:  Initializing SNTP
00:00:04 rtime_info:  Waiting for system time to be set... (1/10)
22:12:53 rtime_info:  22:12:53
22:12:53 setup: ....  Number of saved stations: 13
22:12:53 setup: ....  current station number: 1
22:12:53 setup: ....  current volume: 12
22:12:53 setup: ....  volume is muted: (from 12)
22:12:53 action: ...  switch to station 1
22:12:54 lastURL: ..  http://jazzblues.ice.infomaniak.ch/jazzblues-high.mp3
22:12:54 StreamTitle: BOBBY RUSH  -  Nighttime gardener
E (7756) sdmmc_cmd: sdmmc_read_sectors_dma: sdmmc_send_cmd returned 0x109
E (7757) diskio_sdmmc: sdmmc_read_blocks failed (265)
E (7757) sdmmc_cmd: sdmmc_read_sectors_dma: sdmmc_send_cmd returned 0x109
E (7763) diskio_sdmmc: sdmmc_read_blocks failed (265)
E (7771) sdmmc_cmd: sdmmc_read_sectors_dma: sdmmc_send_cmd returned 0x109
E (7775) diskio_sdmmc: sdmmc_read_blocks failed (265)
E (7780) sdmmc_cmd: sdmmc_read_sectors_dma: sdmmc_send_cmd returned 0x109
E (7786) diskio_sdmmc: sdmmc_read_blocks failed (265)
E (7966) sdmmc_cmd: sdmmc_read_sectors_dma: sdmmc_send_cmd returned 0x109
E (7966) diskio_sdmmc: sdmmc_read_blocks failed (265)
E (7967) sdmmc_cmd: sdmmc_read_sectors_dma: sdmmc_send_cmd returned 0x109
E (7979) diskio_sdmmc: sdmmc_read_blocks failed (265)
22:12:55 ftpServer:   Ftp server waiting for connection on port 21
E (21386) sdmmc_cmd: sdmmc_read_sectors_dma: sdmmc_send_cmd returned 0x109
E (21386) diskio_sdmmc: sdmmc_read_blocks failed (265)
E (21387) sdmmc_cmd: sdmmc_read_sectors_dma: sdmmc_send_cmd returned 0x109

As I see it, The startup runs normally. The stations and the network csv file have been opened, read and closed - so what is causing those errors?

schreibfaul1, 2022-09-03:
0x109 is a (crc) read error. This is most likely a hardware issue.
It may help to reformat the SD card or use a different one. Are there any contact problems?
You can try lowering the frequency in Setup() ``SD_MMC.begin("/sdcard", true, false, 4000)``

clear-sky, 2023-02-19:
hi 
as far as i see in file common.h sd_mmc pins cant be changed in esp32 section but dont see this  comment in esp32-s3 section-can it be changed?

schreibfaul1, 2023-02-19:
With the ESP32, the pins for the SD_MMC are fixed and cannot be changed.
d0 - pin2, clk - pin 14 and cmd - pin 15.
Unlike the ESP32-S3. The pins can be freely selected here, SD_MMC.setPins(clk, cmd, d0);

closed 2024-02-29 by github-actions[bot]