schreibfaul1/ESP32-MiniWebRadio #116: stream not working

spirit89, 2022-09-06:
Any idea how to make this stream work http://24-7nicheradio.com:8110/stream? 

Thanks Dave

schreibfaul1, 2022-09-07:
the mentioned stream works for me, with VS1053 and I2S DAC. Please set -DCORE_DEBUG_LEVEL=4 in platformio.ini. What is the error message?

spirit89, 2022-09-07:
No error message, just added the stream to the stations file and it's not showing on lcd when i cycle the stations.
is there a special way i have to add a stream?

schreibfaul1, 2022-09-07:
You do that in 3 steps:
1) Using the internet browser, go to the homepage and then to the search page. Usually you can use the Community Radio Browser, which will provide the StreamURL, the station name and often the logo. Alternatively, you can enter a StremURL by hand and test the function on the blue button to the right. Then you enter a station name and the LogoURL (you can find a suitable logo e.g. via Google image search) The logo is automatically scaled to the right size. If everything fits save the logo on the SD card with the arrow up button (Upload to SD)
2) Press the left arrow button (add to list). in the edit window you can add a country (Cy). You can repeat steps 1 and 2 several times, the stations will be added at the end of the station list.
3) Once the channel list has been edited, it must still be written to the SD card. To do this, use the green 'Save' button (Save to SD)

If the station list is very long (max. 999 stations are possible), please save the list as a backup in Excel format (blue button 'save xlsx') on your hard drive

spirit89, 2022-09-07:
Are the blue and green buttons in the  Community Radio Browser because i don't see them?

schreibfaul1, 2022-09-07:
the green button is in the list view
[image]

closed 2022-09-08 by spirit89