schreibfaul1/ESP32-MiniWebRadio #117: flac playback

valera39, 2022-09-07:
Hi Wolle!
One configuration is esp32-12k and VS1053, the other is esp32 wower and VS1053.
Actual versions of radio and libraries. There is no playback of flac, mp4a, wav files, aac and mp3 files are played normally.

schreibfaul1, 2022-09-07:
Yes, you are correct. The Vs1053 does not support FLAC out of the box. A patch is also required for this. Unfortunately the patch does not work with all boards. On some boards, the DERQ pin is no longer set and the board stops functioning. So I disabled the feature. You can test if the patch works with your VS1053 board. Find the function in vs1053_ext.cpp "void VS1053::begin()" and you find in the end   // loadUserCode();   just remove the comment marks

valera39, 2022-09-08:
Thank you!
I removed the comments from the line that you indicated and Flac files began to play.
But I still can’t dope WAV files, but for me this is important

closed 2023-05-15 by stale[bot]