schreibfaul1/ESP32-MiniWebRadio #119: Is it internal RTC issue?

eksanmark, 2022-09-24:
Hi..
Saw in the demo, its great project. Try to build last night with hardware as follow: micro sd card adapter, VS1053 (tried also with I2S/MAX chip) and ILI9341 with no success. Code is uploaded successfully, common.h had adjusted as required. Black screen, no sound. Check wiring (still in breadboard) for 3 times, all correct as schematic.
Debug on Serial monitor as "rst:0x10 (RTCWDT_RTC_RESET), boot:0x3b (SPI_FAST_FLASH_BOOT)
flash read err, 1000, ets_main.c 371 ".
Any clue what possible I'm doing wrong?
Thanks.

schreibfaul1, 2022-09-25:
This looks like a problem with the used ESP32 board,, https://www.esp32.com/viewtopic.php?t=25729

eksanmark, 2022-09-25:
Appreciate for your respond, also the link. Finally got it work. The problem is in the VS1053, had used a different type, the red one with straight pin. Not knowing yet what does the root cause, since this part is okay when used with basic code.   
This is the best web radio software I've ever built, thanks for the amazing code.

schreibfaul1, 2022-09-25:
I think pin12 will be the problem, maybe a pull-up resistor on the VS1053 board. If you swap two GPIOs with each other, the error could be eliminated

eksanmark, 2022-09-26:
Yes you're right, I change GPIO 12 to GPIO 08, rearrange the code, upload, and it's working fine, maybe GPIO 12 is RTC pin.
I add some stations via FileZilla including its logo as per example format (writing format, .jpg pixel size and correct name), but the system doesn't yet to read the SD card update, or even when I remove (*) mark for 3 stations in original code, still the same, 13th is the last. Is there something improperly?

ichebyki, 2023-03-19:
You can fuse needed flash voltage and gpio12 will be ignored while boot.

eksanmark, 2023-03-20:
At that time, I follow suggested option that I didn't pay attention before, to add 1k Ohm resistor between SCK and GND, and its work. VS1053 has better sound quality.

closed 2024-02-28 by github-actions[bot]