schreibfaul1/ESP32-MiniWebRadio #121: There are no requests from stations

samand587, 2022-10-18:
Here is my log. There are no requests from stations . What do I need to see?

=~=~=~=~=~=~=~=~=~=~=~= PuTTY log 2022.10.17 21:22:49 =~=~=~=~=~=~=~=~=~=~=~=
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13864
load:0x40080400,len:3608
entry 0x400805f0
ESP32 Chip: ESP32-D0WDQ6
Arduino Version: 2.0.5
ESP-IDF Version: 4.4.2
ARDUINO_LOOP_STACK_SIZE 8192 words (32 bit)

00:00:00 
00:00:00        ***************************    
00:00:00        *     MiniWebRadio V2     *    
00:00:00        ***************************    
00:00:00 
00:00:00 setup: ....  Arduino is pinned to core 1
00:00:00 tft_info: .  init ILI9341
00:00:00 setup: ....  Init SD card
00:00:00 setup: ....  SD card found
00:00:00 setup: ....  seek for stations.csv
00:00:00 setup: ....  stations.csv found
00:00:00 setup: ....  seek for WiFi networks
00:00:02 WiFI_info:   Connecting WiFi...
00:00:02 setup: ....  connected to dlink, IP address is 192.168.0.107
00:00:02 ftpServer:   Buffers allocated: 4622 bytes
00:00:02 rtime_info:  Initializing SNTP
00:00:02 rtime_info:  Waiting for system time to be set... (1/10)
00:00:04 rtime_info:  Waiting for system time to be set... (2/10)
21:22:56 rtime_info:  21:22:56
21:22:56 setup: ....  Number of saved stations: 1
21:22:56 setup: ....  current station number: 1
21:22:56 setup: ....  current volume: 5
21:22:56 VS1053 chipID = -773390214, version = 3
21:22:56 action: ...  current volume is 5
21:22:56 action: ...  switch to station 1
21:22:57 ftpServer:   Ftp server waiting for connection on port 21
21:22:57 icy-descr:   Stream transcoder based on liquidsoap
21:22:57 StationName: europaplus
21:22:57 icy-url: ..  http://95.213.143.165:8010/europaplus320.mp3
21:22:57 lastURL: ..  http://ep256.hostingradio.ru:8052/europaplus256.mp3
21:22:57 StreamTitle: 
21:23:09 action: ...  switch to station 1
21:23:24 action: ...  switch to station 1
21:23:42 action: ...  switch to station 1
21:23:54 action: ...  switch to station 1
21:24:04 action: ...  switch to station 1
21:24:16 AUDIO_info:  Request http:// failed!
21:24:16 StreamTitle: 
21:24:25 action: ...  switch to station 0
21:24:25 AUDIO_info:  Request http:// failed!
21:24:26 StreamTitle: 
21:24:38 action: ...  switch to station 0
21:24:38 AUDIO_info:  Request http:// failed!
21:24:38 StreamTitle: 
21:24:53 action: ...  switch to station 0
21:24:53 AUDIO_info:  Request http:// failed!
21:24:53 StreamTitle: 
21:25:02 action: ...  switch to station 0
21:25:02 AUDIO_info:  Request http:// failed!
21:25:02 StreamTitle:

samand587, 2022-10-18:
=~=~=~=~=~=~=~=~=~=~=~= PuTTY log 2022.10.18 09:21:09 =~=~=~=~=~=~=~=~=~=~=~=
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13864
load:0x40080400,len:3608
entry 0x400805f0
ESP32 Chip: ESP32-D0WDQ6
Arduino Version: 2.0.5
ESP-IDF Version: 4.4.2
ARDUINO_LOOP_STACK_SIZE 8192 words (32 bit)

00:00:00 
00:00:00        ***************************    
00:00:00        *     MiniWebRadio V2     *    
00:00:00        ***************************    
00:00:00 
00:00:00 setup: ....  Arduino is pinned to core 1
00:00:00 tft_info: .  init ILI9341
00:00:00 setup: ....  Init SD card
00:00:00 setup: ....  SD card found
00:00:00 setup: ....  seek for stations.csv
00:00:00 setup: ....  stations.csv found
00:00:00 setup: ....  seek for WiFi networks
00:00:02 WiFI_info:   Connecting WiFi...
00:00:02 setup: ....  connected to dlink, IP address is 192.168.0.107
00:00:02 ftpServer:   Buffers allocated: 4622 bytes
00:00:02 rtime_info:  Initializing SNTP
00:00:02 rtime_info:  Waiting for system time to be set... (1/10)
09:21:16 rtime_info:  09:21:16
09:21:16 setup: ....  Number of saved stations: 1
09:21:16 setup: ....  current station number: 0
09:21:16 setup: ....  current volume: 2
09:21:16 VS1053 chipID = -773390214, version = 3
09:21:16 action: ...  current volume is 2
09:21:16 action: ...  switch to station 0
09:21:16 AUDIO_info:  Request http:// failed!
09:21:16 StreamTitle: 
09:21:16 ftpServer:   Ftp server waiting for connection on port 21
09:21:45 action: ...  switch to station 0
09:21:46 AUDIO_info:  Request http:// failed!
09:21:46 StreamTitle: 
09:21:53 action: ...  switch to station 0
09:21:53 AUDIO_info:  Request http:// failed!
09:21:53 StreamTitle: 
09:22:18 action: ...  switch to station 0
09:22:18 AUDIO_info:  Request http:// failed!
09:22:18 StreamTitle: 
09:22:23 action: ...  switch to station 0
09:22:23 AUDIO_info:  Request http:// failed!
09:22:23 StreamTitle:

samand587, 2022-10-18:
It's with your sheet

=~=~=~=~=~=~=~=~=~=~=~= PuTTY log 2022.10.18 09:29:10 =~=~=~=~=~=~=~=~=~=~=~=
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13864
load:0x40080400,len:3608
entry 0x400805f0
ESP32 Chip: ESP32-D0WDQ6
Arduino Version: 2.0.5
ESP-IDF Version: 4.4.2
ARDUINO_LOOP_STACK_SIZE 8192 words (32 bit)

00:00:00 
00:00:00        ***************************    
00:00:00        *     MiniWebRadio V2     *    
00:00:00        ***************************    
00:00:00 
00:00:00 setup: ....  Arduino is pinned to core 1
00:00:00 tft_info: .  init ILI9341
00:00:00 setup: ....  Init SD card
00:00:00 setup: ....  SD card found
00:00:00 setup: ....  seek for stations.csv
00:00:00 setup: ....  stations.csv found
00:00:00 setup: ....  seek for WiFi networks
00:00:02 WiFI_info:   Connecting WiFi...
00:00:02 setup: ....  connected to dlink, IP address is 192.168.0.107
00:00:02 ftpServer:   Buffers allocated: 4622 bytes
00:00:02 rtime_info:  Initializing SNTP
00:00:02 rtime_info:  Waiting for system time to be set... (1/10)
00:00:04 rtime_info:  Waiting for system time to be set... (2/10)
00:00:06 rtime_info:  Waiting for system time to be set... (3/10)
09:29:21 rtime_info:  09:29:21
09:29:21 setup: ....  Number of saved stations: 1
09:29:21 setup: ....  current station number: 1
09:29:21 setup: ....  current volume: 2
09:29:21 VS1053 chipID = -773390214, version = 3
09:29:21 action: ...  current volume is 2
09:29:21 action: ...  switch to station 1
09:29:22 icy-descr:   Stream transcoder based on liquidsoap
09:29:22 StationName: europaplus
09:29:22 icy-url: ..  http://95.213.143.165:8010/europaplus320.mp3
09:29:22 lastURL: ..  http://ep256.hostingradio.ru:8052/europaplus256.mp3
09:29:22 ftpServer:   Ftp server waiting for connection on port 21
09:29:22 StreamTitle: 
09:29:36 action: ...  switch to station 1
09:29:37 icy-descr:   Stream transcoder based on liquidsoap
09:29:37 StationName: europaplus
09:29:37 icy-url: ..  http://95.213.143.165:8010/europaplus320.mp3
09:29:37 lastURL: ..  http://ep256.hostingradio.ru:8052/europaplus256.mp3
09:29:37 StreamTitle: 
09:29:50 action: ...  switch to station 1
09:29:50 icy-descr:   Stream transcoder based on liquidsoap
09:29:50 StationName: europaplus
09:29:50 icy-url: ..  http://95.213.143.165:8010/europaplus320.mp3
09:29:50 lastURL: ..  http://ep256.hostingradio.ru:8052/europaplus256.mp3
09:29:50 StreamTitle: 
09:29:58 action: ...  switch to station 1
09:29:58 icy-descr:   Stream transcoder based on liquidsoap
09:29:58 StationName: europaplus
09:29:58 icy-url: ..  http://95.213.143.165:8010/europaplus320.mp3
09:29:58 lastURL: ..  http://ep256.hostingradio.ru:8052/europaplus256.mp3
09:29:58 StreamTitle:

closed 2022-10-18 by samand587

schreibfaul1, 2022-10-19:
the link leads to nowhere http://95.213.143.165:8010/europaplus320.mp3
I found this url https://hls-01-regions.emgsound.ru/11_msk/playlist.m3u8 (homepage is https://europaplus.ru/)
found an error in the processing of the TS PES header, please update the audioI2S or the VS1053ext library

samand587, 2022-10-19:
Thanks! There was a problem with the .csv file. Figured it out- great radio!