schreibfaul1/ESP32-MiniWebRadio #122: Connectors 3.5 TFT

Beppi4U, 2022-11-20:
[image]

Hi! Found a suitable TFT 3.5 for the project! Seems that the sd card connectors have different names! 
On your schematic D0 , CMD,CLK in required. On my tft board only miso mosi and cs is shown. Are they comparable? On Display side instead  of TFT_DC  i have T_DO left! Is that right or is'nt the display working with the project? Thanks for helping! Andi

Beppi4U, 2022-11-20:
Thanks for super quick response!  Just to get clear 'shorten out' means soldered out and bridged!?

schreibfaul1, 2022-11-20:
The series resistors are ok for SD(SPI). But in my experience there are problems with SD_MMC. A simple possibility is to bridge the resistors with a wire. In this way, the original state can be restored later

Beppi4U, 2022-11-21:
Hmmm.... Build it yesterday! 
Settings: 
Decoder to VS1053 
TFT and Touch Controller to iLi9488 
Changed the Wifi Credentials to mine .... Compiled fine .....
But....

[image]

Seems theres something wrong with SDCard.
Tried out some other cards ...no success. 
Fat or ExtFat on SD should do the job?
Now i have to check my cables again.

Beppi4U, 2022-11-21:
Why tft_nfo tells me that iLi9341 is initiated when i set it to iLi9488 ?  ;-)

schreibfaul1, 2022-11-21:
Looks like the connection to the SD card is bad, did you use a breadboard?

Beppi4U, 2022-11-22:
No breadboard builds...i prefer DuPont Stecker ! So....i checked the Miso Mosi Sck cable paths from esp pins through SD card adapter twice! Ok! Then measured 3.3 v for SD card...should be good. Maybe Esp32 pinout is not the same as google showed me. [image]
I have used the node mcu ESP3S V1.1   On that Board tft_irq -> pin 39 should be pin svn (grey cable) . Dreq -> pin 36 should be pin svp(white cable) . Can you confirm so far? 
Again thanx for help! Andi

Beppi4U, 2022-11-22:
[image]

Beppi4U, 2022-11-22:
Quick hook up the Card reader...but i guess thats not how it works because now you need chip select!?[image]

schreibfaul1, 2022-11-22:
The wiring looks good. Better to use the display's card adapter, the Catalex adapter has level converters and may not work with SD_MMC.
The CS pin remains free with 'one wire', this is D3 with 'four wire'

Try if the card works at all
````
// ESP32-S3
#include "Arduino.h"
#include "SD_MMC.h"
#include "FS.h"

#define SD_MMC_CMD 14
#define SD_MMC_CLK 13
#define SD_MMC_D0  11

void listDir(fs::FS &fs, const char * dirname, uint8_t levels){
    Serial.printf("Listing directory: %s\n", dirname);

    File root = fs.open(dirname);
    if(!root){
        Serial.println("Failed to open directory");
        return;
    }
    if(!root.isDirectory()){
        Serial.println("Not a directory");
        return;
    }

    File file = root.openNextFile();
    while(file){
        if(file.isDirectory()){
            Serial.print("  DIR : ");
            Serial.println(file.name());
            if(levels){
                listDir(fs, file.path(), levels -1);
            }
        } else {
            Serial.print("  FILE: ");
            Serial.print(file.name());
            Serial.print("  SIZE: ");
            Serial.println(file.size());
        }
        file = root.openNextFile();
    }
}

void setup() {
    Serial.begin(115200);

    pinMode(SD_MMC_D0, INPUT_PULLUP);
    SD_MMC.setPins(SD_MMC_CLK,SD_MMC_CMD, SD_MMC_D0);
    if(!SD_MMC.begin( "/sdmmc", true)){
        Serial.println("Card Mount Failed");
        return;
    }
    listDir(SD_MMC, "/", 0);
}

void loop(){
}
````
and if that's too fast: 
    if(!SD_MMC.begin( "/sdmmc", true, false, 20000)){  //20KHz

Beppi4U, 2022-11-23:
OK! I changed in this little sketch the pin for  CMD CLK and D0 and output is as followed....  ;-)

[image]

Beppi4U, 2022-12-01:
HEUREKA !!! Es läuft! 
Schuld am ganzen Dilemma war zu viele offene Projekte auf einmal in VS Code!  
Schätze da wusste die IDE nicht ganz welche Dateien sie nehmen sollte. 
Alle Projekte entfernt und das richtige geladen .... compilieren und fertig!
Jetzt werde ich mich mal drann machen ein schönes Gehäuse zu basteln! 
Danke für jede Hilfe !
Gruß
Andi

closed 2022-12-01 by Beppi4U

schreibfaul1, 2022-12-01:
Sehr gut!
vG
Wolle

Beppi4U, 2022-12-03:
HEUREKA....es lebt!

[Uploading p4{5440991} (23c0b8ed-ff6c-4310-a7d2-5d7a9a3360d5 id=0x7f0a0400 tag=BaseCommentFragment) …]()

Beppi4U, 2022-12-03:
[image]