schreibfaul1/ESP32-MiniWebRadio #123: SD Card Mount Failed

rodo2021, 2022-11-27:
Hi,
I get the following:

00:00:00        ***************************    
00:00:00        *     MiniWebRadio V2     *    
00:00:00        ***************************    
00:00:00
00:00:00 setup: ....  Arduino is pinned to core 1
00:00:00 tft_info: .  init ILI9486
00:00:00 setup: ....  Init SD card
E (398) sdmmc_sd: sdmmc_init_sd_scr: send_scr (1) returned 0x109
E (398) vfs_fat_sdmmc: sdmmc_card_init failed (0x109).
[   403][E][SD_MMC.cpp:138] begin(): Failed to initialize the card (0x109). Make sure SD card lines have pull-up resistors in place.
00:00:00 SD Card Mount Failed

SD_MMC.begin("/sdcard", true, false, 4000) the change did not help.
The wiring looks good.

Any idea?

schreibfaul1, 2022-11-27:
Reliable contacts and short wires are important. Reducing the clock frequency from 40MHz to 20 or 10MHz often helps.

Philippe-33, 2023-01-26:
Hi rodo2021,
I had the same problem. After searching  about the pinout of sd cards, i have seen that there is 2 gnd pads. I have shorted the two gnd pads and this has solved my problem.
I hope this will help you.

ichebyki, 2023-03-19:
You need pullup all used sd card's pins with 10k resistors. This espressif requirement from datasheet.

clear-sky, 2023-04-02:
hi all
as far as i noticed most of users have problem on mounting sd card
some problems are related to long wires and some on sd card brand and some on sd capacity
if possible please add support native usb port in esp32-s3 modules
in my opinion this is a definite solution

berth59, 2023-05-29:
soldering pin 3 on the sd-card to ground solved my problem with cards.  not recognized

steve6375, 2023-06-08:
I tested with a Mini TF card in the card holder mounted behind my ILI9342 TFT - it just needed 3 wires (MISO, SCLK, MOSI). 
No pullups or anything else required.

I also tested using an Mini-to-Micro SD card adapter + a MicroSD card. I directly soldered onto the minicard adapter contacts. Again no pullups needed. **VCC was 3.3V taken from esp board.** TF\SD cards are 3.3V devices.

However, I did connect both GND contacts and a 1K pullup between 3.3V and CS but I dont think this was needed as it worked fine without the pullup.
Not sure why someone else said to connect CS to GND because that wont work!

[image]

RobinMosedale, 2024-02-03:
Several standalone cards have been attempted, both at 3.3v, 5V, after discovering this failure on the ILI7934.
Changed speed from 40,000,000 HZ, to 20,000,000 Hz, the 10,000,000 Hz.
Finally strapped the CS to 3.3v with 1k.

Solid failure.

Do tell us the secret, or let's just modify it to use SPI on the non critical SPI bus. My passes the usual SPI tests.

Shame, as has much potential

schreibfaul1, 2024-02-03:
Then Espressif's example won't work either. https://github.com/espressif/arduino-esp32/tree/master/libraries/SD_MMC

[image]

RobinMosedale, 2024-02-04:
[image]
Thank you Schreibfaul for both responding so quickly and taking the time to advise.

Context: I'm investigating the BBC's introduction of HLS and MPEG Dash following the removal of the more popular Shout Out. Your suite seemed pretty clear to employ as a test harness. 

However, I've done all you have advised, you can see that the ILI79431 SD card and the standalone MMC adapter have been abandoned with their unwanted resistors. Direct wiring, first without CS pulled up, and secondly with the CS pulled up via 1k ohms to 3.3volts.
The result is the same. I've tried the espressif SDMCC tests both time and basic test.
The espressif tests respond without any message other than logging on to the time site.

I'll move on, and try elsewhere.

Finally, thank you again for your efforts

Robin

Arne65, 2024-02-04:
Hello Robin 

Keep D0 with 10k on 3.3 V actually always worked with the ESP 32

RobinMosedale, 2024-02-04:
Thank you Arne,

Just attempted. No success.
Shame as the card was successfully tested in SPI

Arne65, 2024-02-04:
Set the SD frequency to 1 MHz, the card should then be read and work your way up with the frequency

steve6375, 2024-02-13:
10k not 1k.

kerat71, 2024-02-13:
actually the example from the espressif SDMMC library says 1k, but I think it's not that important, the important thing is that it worked!

Arne65, 2024-02-13:
I think there are different statements regarding the PullUp.
The bottom line is that this is actually due to the level conditions. If in doubt, you have to experiment a bit with the PullUp as not all card adapters are really the same. Some work with 10 K others with 2.2 or 1 K.

Arne65, 2024-02-13:
Hello kerat71

If the card is running you should increase the frequency and see when it can no longer be read safely.

kerat71, 2024-02-13:
Yes, I manage to increase the frequency to 40 MHz and it boots. Then add all the rest hardware (VS1053, TFT) and then it went into bootloop because lack of PSRAM :-( On this ESP32 MiniWebRadio plays almost 5 years.

Arne65, 2024-02-13:
env:esp32]
        board_build.partitions = miniwebradio4MB.csv    ; 4MB Flash
        ;board_build.partitions = miniwebradio8MB.csv    ; 8MB Flash
        ;board_build.partitions = miniwebradio16MB.csv   ; 16MB Flash
        board = ESP32-Dev-4MB                   ; 4MB Flash  *2)
        ;board = ESP32-Dev-8MB                   ; 8MB Flash  *2)
        ;board = ESP32-Dev-16MB                  ;16MB Flash  *2)
build_flags =
        -DCORE_DEBUG_LEVEL=3                    ; 0 None, 1 Error, 2 Warn, 3 Info, 4 Debug, 5 Verbose
        -DCONFIG_ARDUHAL_LOG_COLORS
       // -DBOARD_HAS_PSRAM         Delete this in platformio.ini

There was a time when it had to be deleted if you didn't have PSRAM

closed 2024-03-30 by github-actions[bot]