schreibfaul1/ESP32-MiniWebRadio #124: ILI9488 display with a capacitive touchpad....

oxygeniust, 2022-12-07:
Hello dear schreibfaul1!!I put together your project, tested it and was very satisfied.Today I received the ILI9488 display with a capacitive touchpad.I tried to connect it-there is an image, but the capacitive touch does not work....tell me how you can make it work?

oxygeniust, 2022-12-07:
the display is red,but the pin designations are different

oxygeniust, 2022-12-08:
hello......???????

schreibfaul1, 2022-12-08:
I have no idea how this display works. The display should have a touchpad controller HX8347D. Is it possible that in your case an analog voltage value is returned?

valera39, 2022-12-08:
Perhaps this display has another touchscreen controller

oxygeniust, 2022-12-08:
display ILI9488,touchpad GOODIX GT911...

valera39, 2022-12-08:
Perhaps I'm mistaken, but the radio does not support the display with capacitive touchscreen on GT911.
Common.h prescribes the support of resistive displays with other controllers. Try to set different values #define tp_version, I had a similar problem of incompatibility of the touchscreen and display controller.
Or contact a respected author of the project, for supporting your display.

AVRsteffen, 2023-01-22:
Hello schreibfaul1
Any news on support for the GT911 capacitive touch controller? I have now also ordered such a display and would like to use it with your web radio.

AVRsteffen, 2023-02-20:
I did add stripped down driver only for capacitive touch controller GT911 for 480x320px TFT to tft.h and tft.cpp files. I can't upload here but can show you that additional code segments.

**tft.h**
`#include "Wire.h"

// TFT code here

// Touch code here
//-----------------------------------------------------------------------------------------------------------------------

struct __attribute__((packed)) GTInfo {
  char productId[4]; // 0x8140 - 0x8143 
  uint16_t fwId; // 0x8144 - 0x8145 
  uint16_t xResolution; // 0x8146 - 0x8147
  uint16_t yResolution; // 0x8148 - 0x8149
  uint8_t vendorId; // 0x814A
};

struct __attribute__((packed)) GTPoint {
  // 0x814F-0x8156, ... 0x8176 (5 points) 
  uint8_t trackId;
  uint16_t x;
  uint16_t y;
  uint16_t area;
  uint8_t reserved;
};

// 0xBA/0xBB (0x5D 7bit)
#define GT911_I2C_ADDR_BA     0x5D
#define GT911_REG_DATA        0x8140
//#define GT911_REG_ID 0x8140
#define GT911_REG_COORD_ADDR  0x814E

class GT911 {
  public:
    enum class Rotate {
      _0,     // normal
      _1,     // normal invers
      _2,     // 180 deg
      _3,     // 180 deg invers
    };
  private:
    TwoWire *_wire;
    uint8_t _addr;
    uint16_t x=0, y=0;

    //bool _configLoaded = false;
    boolean f_loop = true;
    GTInfo _info;
    GTPoint _points[0];
    uint8_t _rotation;

    //void reset();
    void i2cStart(uint16_t reg);
    bool write(uint16_t reg, uint8_t data);
    uint8_t read(uint16_t reg);
    bool writeBytes(uint16_t reg, uint8_t *data, uint16_t size);
    bool readBytes(uint16_t reg, uint8_t *data, uint16_t size);
    bool readTouches();
    bool readTouchPoints();
  public:
    GT911(TwoWire *twi = &Wire);
    bool begin(uint8_t addr = GT911_I2C_ADDR_BA, uint32_t clk = 400000);
    GTInfo* readInfo();
    GTPoint *getPoints();

    void setRotation(uint8_t rotation);
    void loop();
};
`

Changes in **tft.ccp**
`#include <Wire.h>

/***********************************************************************************************************************
                                            T O U C H P A D
***********************************************************************************************************************/

// Code für capacitive Touchpad mit GT911
GT911::GT911(TwoWire *twi) : _wire(twi ? twi : &Wire) {

}

void GT911::i2cStart(uint16_t reg) {
  _wire->beginTransmission(_addr);
  _wire->write(reg >> 8);
  _wire->write(reg & 0xFF);
}

bool GT911::write(uint16_t reg, uint8_t data) {
  i2cStart(reg);
  _wire->write(data);
  return _wire->endTransmission() == 0;
}

uint8_t GT911::read(uint16_t reg) {
  i2cStart(reg);
  _wire->endTransmission();
  _wire->requestFrom(_addr, (uint8_t)1);
  while (_wire->available()) {
    return _wire->read();
  }
  return 0;
}

bool GT911::writeBytes(uint16_t reg, uint8_t *data, uint16_t size) {
  i2cStart(reg);
  for (uint16_t i = 0; i < size; i++) {
    _wire->write(data[i]);
  }
  return _wire->endTransmission() == 0;
}

bool GT911::readBytes(uint16_t reg, uint8_t *data, uint16_t size) {
  i2cStart(reg);
  _wire->endTransmission();

  uint16_t index = 0;
  while (index < size) {
    uint8_t req = _min(size - index, I2C_BUFFER_LENGTH);
    _wire->requestFrom(_addr, req);
    while (_wire->available()) {
      data[index++] = _wire->read();
    }
    index++;
  }

  return size == index - 1;
}

bool GT911::readTouches() {
  uint32_t timeout = millis() + 20;
  do {
    uint8_t flag = read(GT911_REG_COORD_ADDR);
    if (flag & 0x80) {
      write(GT911_REG_COORD_ADDR, 0);
    return true;
    }
    delay(1);
  } while (millis() < timeout);

  return false;
}

bool GT911::readTouchPoints() {
  bool result = readBytes(GT911_REG_COORD_ADDR + 1, (uint8_t*)_points, sizeof(GTPoint));

  if (result) {
    //for (uint8_t i = 0; i < GT911_MAX_CONTACTS; i++) {
      if (_rotation == 0) { _points[0].y = _info.yResolution - _points[0].y; }
    //  if (_rotation == 1) { _points[0].y = _info.yResolution - _points[0].y; }
      if (_rotation == 2) { _points[0].x = _info.xResolution - _points[0].x; }
      if (_rotation == 3) { _points[0].x = _info.xResolution - _points[0].x;
                            _points[0].y = _info.yResolution - _points[0].y; }
    }
  //}
  return result;
}

bool GT911::begin(uint8_t addr, uint32_t clk) {
  _addr = addr;
  
  delay(500);
 
  _wire->begin();
  _wire->setClock(clk);
  _wire->beginTransmission(_addr);
  if (_wire->endTransmission() == 0) {
    readInfo(); // Need to get resolution to use rotation
    return true;
  }
  return false;
}

GTInfo* GT911::readInfo() {
  readBytes(GT911_REG_DATA, (uint8_t*)&_info, sizeof(_info));
  return &_info;
}

GTPoint *GT911::getPoints() {
  return _points;
}

void GT911::setRotation(uint8_t rotation) {
  _rotation = rotation;
}

void GT911::loop() {
	
    if (readTouches()) {
      if (f_loop==true) {
            f_loop = false;
            readTouchPoints();
            GTPoint* ts = getPoints();
            y = ts[0].x;
            x = ts[0].y;
            if (tp_pressed) tp_pressed(x, y);
                Serial.printf(" pressed: x=%d , y=%d s:%d\n", x, y, ts[0].area);
      }
    } else {
      if (f_loop == false) {
            if (tp_released) tp_released();
				Serial.printf("released: x=%d , y=%d \n", x, y);
      }
      f_loop = true;
    }
}
//----------------------------------------------------------------------------------------------------------------------
`
Only the two I2C lines are required. TP_INT must be connected in series to GND via a resistor so that TP_INT is pulled low. TP_RES is connected to /EN (RESET) of the ESP32.

AVRsteffen, 2023-02-20:
Changes in **main.cpp**

Line 113:  change to `GT911 tp = GT911();`

In `void setup();
.......
`
Line 1084:  remove `tp.setVersion(TP_VERSION);`  and replace with ` tp.begin() `

closed 2024-02-26 by github-actions[bot]