schreibfaul1/ESP32-MiniWebRadio #127: The flicker of time

valera39, 2022-12-18:
Hi Wolle!
Can you make so that the time on the display does not flicker to the beat every second?
[image]
And one more question. According to your last records in the repository, I realized that DLNA support will be implemented in the Mini Web Radio. It would be very cool!
Or I'm wrong?

schreibfaul1, 2022-12-18:
Hello valera39,
the flickering thing can be fixed for sure. So far I have completely deleted the window with the time and rewritten the time. That's the simple solution. It is better to rewrite the digit that really changes. That will come soon in an update.
DLNA is in the experimental stage. Everything is not going well yet. I can imagine implementing DLNA in the future.

valera39, 2022-12-18:
OK, I will wait for updates, good luck with the implementation of DLNA

closed 2022-12-20 by valera39