schreibfaul1/ESP32-MiniWebRadio #130: Constant reboot

valera39, 2022-12-20:
Hi Wolle!
After the last update, when choosing a station https://fffutu.re/Retro There is a constant reboot
radio. It is not even possible to go to the web interface and choose another stream.
00:00:00    [0m
00:00:00 [33m       ***************************    [0m
00:00:00 [33m       *     MiniWebRadio V2     *    [0m
00:00:00 [33m       ***************************    [0m
00:00:00    [0m
00:00:00 setup: ....  Arduino is pinned to core [36m1[0m
00:00:00 tft_info: .  init [36mILI9488[0m
00:00:00 setup: ....  Init SD card[0m
00:00:00 [37msetup: ....  SD card found[0m
00:00:01 setup: ....  seek for stations.csv[0m
00:00:01 setup: ....  stations.csv found[0m
00:00:01 setup: ....  seek for WiFi networks[0m
[0;32m[  1727][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: ASUS_AC66U[0m
[0;32m[  1741][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: Honor[0m
[0;32m[  1742][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: network_01[0m
[0;32m[  1749][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: network_02[0m
[0;32m[  1757][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: network_03[0m
[0;32m[  1764][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: network_04[0m
[0;32m[  1772][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: network_04[0m
[0;32m[  8250][I][WiFiMulti.cpp:114] run(): [WIFI] scan done[0m
[0;32m[  8251][I][WiFiMulti.cpp:119] run(): [WIFI] 5 networks found[0m
[0;32m[  8252][I][WiFiMulti.cpp:160] run(): [WIFI] Connecting BSSID: 2C:FD:A1:C2:44:F0 SSID: ASUS_A
C66U Channel: 9 (-54)[0m
[0;32m[  8327][I][WiFiMulti.cpp:174] run(): [WIFI] Connecting done.[0m
00:00:07 WiFI_info:   Connecting WiFi...[0m
00:00:07 setup: ....  connected to [36mASUS_AC66U[37m, IP address is [36m192.168.1.133[0m
00:00:07 ftpServer:   Buffers allocated: 4622 bytes[0m
00:00:07 rtime_info:  Initializing SNTP[0m
00:00:07 rtime_info:  Waiting for system time to be set... (1/10)[0m
00:00:09 rtime_info:  Waiting for system time to be set... (2/10)[0m
00:00:11 rtime_info:  Waiting for system time to be set... (3/10)[0m
23:11:50 rtime_info:  23:11:50[0m
23:11:50 audiotask:   is pinned to core [36m1[0m
23:11:50 audiotask:   priority is [36m2[0m
23:11:50 setup: ....  Number of saved stations: [36m82[0m
23:11:50 setup: ....  current station number: [36m71[0m
23:11:50 setup: ....  current volume: [36m2[0m
23:11:51 VS1053 chipID = [36m-1350368918[37m, version = [36m3[0m
23:11:51 action: ...  current volume is [36m2[0m
23:11:52 action: ...  switch to station [36m71[0m
23:11:52 AUDIO_info:  [32mConnect to new host: "https://fffutu.re/Retro"[0m
23:11:52 AUDIO_info:  [32mSong stopped correctly after 0 msec[0m
23:11:52 AUDIO_info:  [32mPSRAM not found, inputBufferSize: 11903 bytes[0m
23:11:52 AUDIO_info:  [32mSSL has been established in 569 ms, free Heap: 156400 bytes[0m
23:11:52 AUDIO_info:  [32mredirect to new host "https://stream.zeno.fm/fs7e5zt06qzuv?*****"[0m
23:11:52 AUDIO_info:  [32mConnect to new host: "https://stream.zeno.fm/fs7e5zt06qzuv?*****"[0m
23:11:52 AUDIO_info:  [32mSong stopped correctly after 0 msec[0m
Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception).
Debug exception reason: Stack canary watchpoint triggered (audioplay)
Core  1 register dump:
PC      : 0x4037fd33  PS      : 0x00060236  A0      : 0x8037e003  A1      : 0x3fcaaa70
A2      : 0x3fcaa990  A3      : 0xb33fffff  A4      : 0x0000cdcd  A5      : 0x00060223
A6      : 0x00060220  A7      : 0x0000abab  A8      : 0xb33fffff  A9      : 0xffffffff
A10     : 0x00060223  A11     : 0x00000000  A12     : 0x00060220  A13     : 0x00000000
A14     : 0x02caa990  A15     : 0x00ffffff  SAR     : 0x00000011  EXCCAUSE: 0x00000001
EXCVADDR: 0x00000000  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0x00000000

Backtrace:0x4037fd30:0x3fcaaa700x4037e000:0x3fcaaab0 0x4037e0c0:0x3fcaaaf0 0x403790dd:0x3fcaab10 0x403791c9:0x3fcaab40 0x42097219:0x3fcaab60 0x42097233:0x3fcaab80 0x42072f8b:0x3fcaaba0 0x420713e1:0x3fcaabc0 0x420714a1:0x3fcaabf0 0x4206fe7e:0x3fcaac10 0x4206b2c1:0x3fcaad20 0x4206b3a3:0x3fcaad90 0x4206b3eb:0x3fcaae40 0x4206b559:0x3fcaaec0 0x42067521:0x3fcaaee0 0x420675f2:0x3fcab090 0x420676c5:0x3fcab0b0 0x4206778d:0x3fcab120 0x42065052:0x3fcab140 0x42065f6a:0x3fcab170 0x420687cc:0x3fcab1c0 0x42069465:0x3fcab200 0x42069bbf:0x3fcab3f0 0x42067cad:0x3fcab550 0x42067ea0:0x3fcab580 0x42062eac:0x3fcab5a0 0x42060c5d:0x3fcab640 0x42060c97:0x3fcab660 0x420a0f49:0x3fcab680 0x420a0089:0x3fcab920 0x420a020a:0x3fcab970 0x420bcbcd:0x3fcab990 0x4201481d:0x3fcab9b0 0x42015341:0x3fcabd50 0x42015685:0x3fcac1d0 0x42015712:0x3fcac1f0 0x42003986:0x3fcac210

ELF file SHA256: 0000000000000000

Rebooting...
�ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x420bb89e
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x43c
load:0x403c9700,len:0xbec
load:0x403cc700,len:0x2a3c
SHA-256 comparison failed:
Calculated: 3611f330726ad4edf64f62982b0568332a5800bc59425cab90be5379db92c108
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x403c98d8
ESP32 Chip: ESP32-S3
Arduino Version: 2.0.5
ESP-IDF Version: 4.4.2
ARDUINO_LOOP_STACK_SIZE 8192 words (32 bit)

And also pay attention to the message #128

schreibfaul1, 2022-12-20:
I have it too. I reduced the stack of the audio task. and that worked wonderfully, only at https://fffutu.re/Retro the stack is no longer sufficient. If you click on the question mark on the web page you can see the 'audiotask .. stackHighWaterMark: ' in the terminal, this is the minimum free value that was reached at some point. When the value goes to zero, ``Debug exception reason: Stack canary watchpoint triggered (audioplay)`` happens. I will restore the old value again. And thanks for reporting.
As for #128, it's my turn. In the first step, no more file names separated by commas are transmitted to the web, but a JSON string. That's the way it is now, maybe it will help. This will be completely revised soon.

valera39, 2022-12-21:
Great job!

closed 2022-12-22 by valera39