schreibfaul1/ESP32-MiniWebRadio #131: Slower graphics build in version 2

AVRsteffen, 2023-02-01:
Since the new V2, the graphic structure of the buttons and other images from the SD card is very slow.  I compared it with V1.  Certainly there was no support from the 480x320 pixel displays.  But even the 320x240s have a very slow graphic structure in V2.  What could be the reason?

schreibfaul1, 2023-02-02:
The difference to version 1 is that the SD card no longer shares the SPI bus with the display. The speed should be at least equal to the V1. The SD_MMC doesn't like long wires or contact problems (or series resistors that are built into some displays). Not every SD card is equally suitable. Good SD cards tolerate 40 or 80MHz, you can change that in common.h.

AVRsteffen, 2023-02-04:
Hey Wolle, I also have the data running via MMC in V1. I was the one who converted this to 2 cards. One SD card for audio mp3 and one MMC for data. I can run both TFT and MMC with 80Mhz in V2 (see video).  Nevertheless, the V2 is much slower in graphics construction than the V1. It also looks like there are always pauses of a few milliseconds between the individual images. Do you have any idea where that could come from?

AVRsteffen, 2023-02-04:
Can't upload video or don't know how to do..

AVRsteffen, 2023-02-04:
Uploading YouCut_20230204_124248087.mp4…

AVRsteffen, 2023-02-04:
Both, V1 and V2 uses 320x240 pixel displays with ILI9341 controller.

https://user-images.githubusercontent.com/2552410/216766812-2636734a-792d-4db0-8116-8f6c8637faad.mp4

schreibfaul1, 2023-02-04:
One cannot expect miracles from displays with SPI. This is my radio with a display from the Raspberry Pi with 480x320px. You can try setting the audio task to core 0.

https://user-images.githubusercontent.com/26044260/216769412-097a9f76-9e54-4dd8-9f91-70dd09acc656.mp4

AVRsteffen, 2023-02-04:
I use external DAC PCM5102A instead of a VS1053 module. Maybe it's because. I had already tried the original 480x320 px RaspberryPi display. It was just as slow. As I said, there seem to be breaks between the individual pictures. I'll try the audio task with the Core 0.

closed 2023-05-15 by stale[bot]

AVRsteffen, 2023-07-29:
The problem is not solved for me yet. How fast is your display in V2 and with which hardware configuration? What I do wrong?

My configuratuon:
• ESP32 WROVER / 4MB Flash / 8MB PSRAM
• DAC PCM5102A
• DISPLAY ILI9488 480x320 pixel @ 40MHz
• SD Card 32GB at 80MHz

Uploading YouCut_20230729_151225011.mp4…

steve6375, 2023-07-29:
Button load speed can be affected by the SD card frequency
Try increasing speed in common.h
We cannot see your settings so we do not know what settings you are using.

AVRsteffen, 2023-07-29:
I can't increase anymore. I use already 80MHz for SD/MMC card. I did try with 40MHz, 20MHz and differend cards to. It's nothing speed up graphics. If speedup TFT frequency to 80MHz I get some wrong graphics. Mainly the buttons are wrong drawed.

What configuration you have?

valera39, 2023-07-29:
try changing AUDIOTASK_CORE to 0 and AUDIOTASK_PRIO to 2

steve6375, 2023-07-29:
See #262

AVRsteffen, 2023-07-29:
Here I will show you:

https://github.com/schreibfaul1/ESP32-MiniWebRadio/assets/2552410/4d44b6f3-932a-4a6e-9a5f-41a9c2d6c092

valera39, 2023-07-29:
What if you also try another card of a higher grade?

AVRsteffen, 2023-07-29:
My common.h

[image]

ESP32 Chip: ESP32-D0WD-V3
Arduino Version: 2.0.11
ESP-IDF Version: 4.4.4
MiniWebRadio Version 2.8.3b Jul 22/2023
CPU speed 240 MHz
ARDUINO_LOOP_STACK_SIZE 8192 words (32 bit)
FLASH size 8388608 bytes, speed 40 MHz
PSRAM total size: 8388608 bytes

00:00:00
00:00:00 ***************************
00:00:00 * MiniWebRadio V2 *
00:00:00 ***************************
00:00:00
00:00:00 setup: .... Arduino is pinned to core 1
00:00:00 tft_info: . init ILI9488
00:00:00 setup: .... Init SD card
00:00:00 setup: .... SD card found, 3762.6 MB by 3781.5 MB free
00:00:01 setup: .... seek for stations.csv
00:00:01 setup: .... stations.csv found
00:00:01 setup: .... seek for WiFi networks
[ 1638][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: mySSID

I am just pinned core in common.h to core --> 0
And I guess TFT runs faster..

AVRsteffen, 2023-07-29:
I am just tested - TFT runs faster with core -> 0 in common.h

What I don't understand is terminals output "Arduino is pinned to core 1" in both configurations of core!

steve6375, 2023-07-29:
I pushed my ILI9341 to use
```
#define TFT_FREQUENCY       70000000                // 27000000, 40000000, 80000000
#define SDMMC_FREQUENCY     70000000          // 40000000, 2000000, 10000000, not every SD Card will run at 40MHz
```

valera39, 2023-07-29:
What if you also try another higher end card?

steve6375, 2023-07-29:
https://randomnerdtutorials.com/esp32-dual-core-arduino-ide/
Arduino uses 1
Audio uses 0
I tested changing to priority 12

```
00:00:00 setup: ....  Arduino is pinned to core 1
00:00:00 tft_info: .  init ILI9341
00:00:00 setup: ....  Init SD card
00:00:00 setup: ....  SD card found, 3381.6 MB by 3886.5 MB free
00:00:01 setup: ....  seek for stations.csv
00:00:01 setup: ....  stations.csv found
00:00:01 setup: ....  seek for WiFi networks
..
17:06:08 rtime_info:  17:06:08
17:06:08 audiotask:   is pinned to core 0
17:06:08 audiotask:   priority is 12

```

Jefke1, 2023-07-31:
> Here I will show you:
> 
>  YouCut_20230729_151225011.mp4

What strikes me is not the speed. But your VU meter the color is white with me when in action it is colored and the top part is darker.

AVRsteffen, 2023-07-31:
This is only a effect of camera.