schreibfaul1/ESP32-MiniWebRadio #136: Audio Player Loop button

ve2cfb, 2023-03-10:
Hello, my name is Pierre and i love your project.
Had a bit of problem with the SD card but now everything is under control.
Would it be possible to add a button in the Audio Player section (after the Blue_Ready_Button)  and have the mp3 music files loop continuously.  It could be an ON/OFF type button.
Thanks again...

ve2cfb, 2023-03-13:
****Hello** again,
**Thanks for the tip about Playlist.
I had problems with mp3 files location and M3U files but everything is working fine now(***almost).

****Under "audiofiles", i created 1 directory  "musique".
Under "audiofiles", created 3 M3U files 
Under "Musique", made 3 directories, "medley", "classique", "rieu" and each have mp3 files.
===============================================**
**Here is an example of one M3U FILE:**
#EXTM3U
#EXTINF:140,Guy Lombardo - Boo Hoo
musique/medley/Boo.MP3
#EXTINF:315,Creedence Clearwater Revival - Born On The Bayou
**musique/medley/BornOnTheB.mp3**
#EXTINF:161,Mama's & Papa's - California Dreamin'
musique/medley/CaliforniaDrea.mp3
========================================**
*** with this arrangment, in the Audio Player, it does not show the M3U file, 
you have to hit the right button one time in order to see the first M3U.
 The following thing is the %20 and other character that was adressed in one post. But i can live with that.

Many thanks again...  Pierre****

schreibfaul1, 2023-03-13:
Saving audio files in subfolders makes sense for m3u. I've changed that now. I created the folder structure with FileZilla.
[image]
Your test playlist is now running.
PS Since Arduino V2.0.6, UTF8 filenames are also allowed and %20 is not necessary.

ve2cfb, 2023-03-14:
Hi again,

Concerning the special character here is my problem.

Here is what VLC  did generate...
================  Does not work  ==========================
#EXTM3U
#EXTINF:207,Edith Piaf - Hymne À L'amour
musique/medley/05%20Hymne%20%C3%80%20L'amour.mp3
=====================================================

Here are the mods i made.
================  This work ok  ==========================
#EXTM3U
#EXTINF:207,Edith Piaf - Hymne À L'amour
musique/medley/05 Hymne A L'amour.mp3
================================================== 
All i did on the mp3 file is change the "À" to an "A" and in the M3U, remove all the reference to %xx%xx

Pierre