schreibfaul1/ESP32-MiniWebRadio #137: AUDIO_info:  Metadata block to long! Skipping all Metadata from now on. Causes a sound problem: BR Radio only!

TubaFonse, 2023-03-11:
Hello Wolle, 
due to your great prepared project and good documentation now I successfully built my 1st own MiniWebRadio V2 (ESP32-DevKitC & ext. VS1053-Module). 
But since a few days I got continuous sound distortions (bad "plopp's" and "quieks" from the speakers), but only when I listen any of the 14 different livestreams of "BR Radio (Bayrischer Rundfunk)". e.g. url's: https://streams.br.de/bayern1obb_2.m3u 
I found the reason of the problem via PlatformIO Serial Monitor: 

AUDIO_info:  Metadata block to long! Skipping all Metadata from now on. 

I found, that all the stations of BR Radio send (~since a few days?) a longer Metadata-block, than expected/allowed in your library vs1053_ext.cpp (see line No. 3268): 

if(!metalen) {
        int b = _client->read();   // First byte of metadata?
        metalen = b * 16 ;                              // New count for metadata including length byte
        if(metalen > 512){
            AUDIO_INFO("Metadata block to long! Skipping all Metadata from now on.");
            m_f_metadata = false;                        // expect stream without metadata
            return 1;
        }

To test/try, how much more Metadata Bytes than 512 sends BR Radio from now on, I increased the value 512 + 16, then the sound distortions were finished!!! But I got a txt overflow ...
If you can reproduce / confirm the problem, please help me to fix the problem "professional" by extend the length of Metadata block 
(Due to I'm a very bad programmer = beginner). 

Many thanks, 
da TubaFonse.

schreibfaul1, 2023-03-12:
Hi @TubaFonse,
yes, the metadata exceeds 512 bytes. I copied the 'readMetadata()' routine from the audioI2S library, which neutralizes this error.
What is transmitted as metadata with 'BAYERN 1 Oberbayern' is 'peculiar'
`11:02:59 StreamTitle: yern1.deInternet: www.bayern1.deInternet: www.bayern1.deInternet: www.bayern1.deInternet: www.bayern1.deInternet: www.bayern1.deInternet: www.bayern1.deInternet: www.bayern1.deInternet: www.bayern1.deInternet: www.bayern1.deInternet: www.bayern1.deInternet: www.bayern1.deInternet: www.bayern1.deInternet: www.bayern1.deInternet: www.bayern1.deInternet: www.bayern1.deInternet: www.bayern1.deInternet: www.bayern1.deInternet: www.bayern1.deInternet: www.bayern1.deInternet: www.bayern1.deInternet: www.bayern1.d`
Hope the error is fixed with this.
Thank you for your contribution and best regards
Wolle

TubaFonse, 2023-03-12:
Dear @schreibfaul1 Wolle, 
many thanks for your quick response and fixing the error!!! 
In the meantime I sent an e-mail to "service-team Bayerischer Rundfunk - Internet-Radio" to fix the 'peculiar' BR-metadata ;-)
I don't believe, that the BR-team will be as quick as you in solving problems ... we will see. 
Have a nice sunday! 
Best regards, 
da TubaFonse

closed 2023-03-12 by TubaFonse