schreibfaul1/ESP32-MiniWebRadio #139: Support of the Cyrillic alphabet

valera39, 2023-03-15:
Hello Wolle!
I am using version 2.5 of your radio, is it possible to make Cyrillic support in the names of files and playlists for an audiopler?
And yet, the "back" button in the audio player returns to the beginning of the file list, and not to the previous file. It can be fixed?
Thank you!

schreibfaul1, 2023-03-15:
Hello valera39,
yes, this is now possible. Audio files can contain Cyrillic letters. This also applies to image files that contain the logo. This makes it possible to save station names in the 'original', without an |ASCII name for the logo.
I adjusted the 'Content_on_SD_Card.zip', there you can see what I mean (stations.csv).
In addition, subfolders for playlists can be created within the 'audiofiles' folder. These folders are not visible, but a large number of audio files can now be saved that can be played via an m3u playlist. Creating a playlist is very easy with VLC Player. There is also an example for this in the .zip

Espressif offers the openNextFile() command for managing SD cards. This is good to use. Unfortunately there is no openPreviousFile()   :-(

valera39, 2023-03-15:
Thanks for the quick response!
When trying to upload files with Cyrillic on FTP, Filezilla writes "The server does not support the symbols not ascii."
I managed to copy files with Cyrillic only by inserting the SD card into the computer, but when you repeatedly connect via FTP, Filezilla does not correctly display the names, instead of
"Logo-2" writes "Р›РѕРіРѕ-2 ". It can be fixed?

schreibfaul1, 2023-03-15:
I forgot to mention this:
in FileZilla the character set must be changed from 'Use custom charset' to 'Force UTF-8'.

schreibfaul1, 2023-03-15:
so that the directory tree in FileZilla looks like this:
[image]
in Files/Servermanager/Charset set UTF-8
[image]
Sorry, this is in German, but you'll find the setting for sure