schreibfaul1/ESP32-MiniWebRadio #140: FLAC on ESP32-S3 from Espressif

valera39, 2023-03-15:
Hello Wolle!
Configuration: ESP32-S3 and PCM5102.

19:00:41 AUDIO_info:  [32mConnect to new host: "http://89.223.45.5:8000/nota-flac"[0m
19:00:41 AUDIO_info:  [32mbuffers freed, free Heap: 146068 bytes[0m
19:00:41 AUDIO_info:  [32mConnection has been established in 39 ms, free Heap: 145476 bytes[0m
19:00:41 icy-descr:   Classics forever![0m
19:00:41 AUDIO_info:  [32micy-name: SECTOR | Nota channel[0m
19:00:41 StationName: [35mSECTOR | Nota channel[0m
19:00:42 icy-url: ..  www.sectorradio.com[0m
19:00:42 AUDIO_info:  [32mFLAC works only with PSRAM![0m

As I understand it, there will be no reproduction without psram?
Is it possible to make FLAC support on ESP32-S3 from Espressif without PSRAM?
Thank you!

valera39, 2023-03-16:
loadUserCode() Activated earlier, but it does not help

schreibfaul1, 2023-03-17:
Too bad, I have two boards from the same manufacturer and FLAC only works on one. Basically, it doesn't work at all without a patch. There are different patches for VLSI, maybe something else works.http://www.vlsi.fi/en/products/vs1053.html
Basically, I would not rely on this old technology anymore, rather on I2S and an external DAC.

valera39, 2023-03-17:
But I also produced PCM 5102 with different ESP32, but FLAC did not work. What is the best configuration in your opinion?