schreibfaul1/ESP32-MiniWebRadio #142: MultiWifi Help

ve2cfb, 2023-03-23:
Hello Wolle,
If i have my normal credentials in the common.h everything works ok.
When I tried using the multi wifi in the network.csv, i cannot connect..
I followed your instruction about using TAB when creating SSID and Password.

Here is an output of the Serial Monitor.
In the common.h i changed the ssid and password to an invalid one.
In the network.csv i added my own wifi credential plus an extra one.

//============================================================
***  in networks.csv
*SSID	Password	Info
network_01	Bouboule	   dummy
network_02	COGECO-6d370	dummy
network_03	xxxxx	dummy
network_04	xxxxx	dummy
network_04	xxxxx	dummy
	
		
* MiniWebRadio will scan all available WiFi Networks		
* First try is to connect with given credentials in main.cpp		
* Is this not succesfully MiniWebRadio will scan all available WiFi Networks		
* then connect with the given entries in the table		
*		
* empty lines will be skipped, lines starts with * are ignored		
* tabs are delimitter, use tabs between columns, no whitespaces		
* eg.   SSID{TAB}Password{TAB}Info{ENTER}		
* 		
//=========================================================			
****  In common.h

// updated: 14.Mar.2023

#pragma once

#define _SSID               "AAABouboule"                        // Your WiFi credentials here
#define _PW                 "dummy"
#define TZName              "EST5EDT,M3.2.0,M11.1.0"    // Timezone (more TZNames in "rtime.cpp")
#define DECODER             0                               // (0)VS1053 , (1)MAX98357A PCM5102A... (2)AC101 (3)ES8388 (4)WM8978
//==============================================================================

---- Opened the serial port COM3 ----
[  7471][V][WiFiGeneric.cpp:383] _arduino_event_cb(): SCAN Done: ID: 128, Status: 0, Results: 10
[  7472][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 1 - SCAN_DONE
[  7478][I][WiFiMulti.cpp:114] run(): [WIFI] scan done
[  7484][I][WiFiMulti.cpp:119] run(): [WIFI] 10 networks found
[  7490][D][WiFiMulti.cpp:151] run():        0: [1][20:6A:94:01:F2:28] Bouboule (-60) *   

//  did find my router during the scan but not in the network.csv 

[  7499][D][WiFiMulti.cpp:151] run():        1: [6][34:5D:9E:EE:53:42] lmnop2.4 (-79) *
[  7508][D][WiFiMulti.cpp:151] run():        2: [1][20:6A:94:01:91:98] COGECO-19190 (-81) *
[  7517][D][WiFiMulti.cpp:151] run():        3: [1][88:57:1D:E9:D7:C3] [range]_E30AJT7113274D (-89) *
[  7526][D][WiFiMulti.cpp:151] run():        4: [1][1C:9D:C2:F6:86:B9] WACFan_F686B8 (-89) *
[  7536][D][WiFiMulti.cpp:151] run():        5: [11][0C:AC:8A:2B:16:61] BELL970 (-89) *
[  7544][D][WiFiMulti.cpp:151] run():        6: [1][84:0B:7C:EC:FE:A8] COGECO-CFEA0 (-91) *
[  7553][D][WiFiMulti.cpp:151] run():        7: [6][34:5D:9E:EF:C9:98] BELL199 (-92) *
[  7562][D][WiFiMulti.cpp:151] run():        8: [11][80:20:DA:0E:CA:64] BELL550 (-92) *
[  7571][D][WiFiMulti.cpp:151] run():        9: [1][A8:6A:BB:EA:BA:DE] BELL260 (-94) *
[  7579][E][WiFiMulti.cpp:191] run(): [WIFI] no matching wifi found!
00:00:10 WiFi credentials are not correct

00:00:11 WiFi credentials are not correct
//======================================================================

Many thanks for your time.

schreibfaul1, 2023-03-23:
WiFiMulti searches for all available networks and lists them.
You can enter several networks + password in the networks.csv, e.g.

lmnop2.4    1234567    this is in my office
COGECO-19190     mySecretPassword
BELL970    PWforBELL970

The csv can also be created with a spreadsheet, it is important that the entries are separated by a TAB
* eg.   SSID{TAB}}Password{TAB}}Info{ENTER}

If you edited a networks.csv, you don't need any credentials in common.h.
You can see the credentials in common.h as the first entry in the csv.

ve2cfb, 2023-03-25:
Hello Wolle,
I made sure the networks.csv was alright.

networks.csv
network_01	COGECO-6D370	D78C7COGECO
network_02	Bouboule	     a4b5c6d7e8

Hex Dump:
6E 65 74 77 6F 72 6B 5F 30 31 09 43 4F 47 45 43 4F 2D 36 44 33 37 30 09 44 37 38 43 37 43 4F 47 45 43 4F 0A 6E 65 74 77 6F 72 6B 5F 30 32 09 42 6F 75 62 6F 75 6C 65 09 61 34 62 35 63 36 64 37 65 38 0A

the common.h file has the following: 

#define _SSID               ""        // Your WiFi credentials here
#define _PW                 ""

In the main.c
i uncomment this line...
            log_i("s_ssid=%s  s_password=%s  s_info=%s", s_ssid.c_str(), s_password.c_str(), s_info.c_str());

here is a serial output of the error.

00:00:00    
00:00:00        ***************************    
00:00:00        *     MiniWebRadio V2     *    
00:00:00        ***************************    
00:00:00    
00:00:00 setup: ....  Arduino is pinned to core 1
00:00:00 tft_info: .  init ILI9488
00:00:00 setup: ....  Init SD card
00:00:00 setup: ....  SD card found
E (269) ledc: ledc_get_duty(739): LEDC is not initialized
00:00:01 setup: ....  seek for stations.csv
00:00:01 setup: ....  stations.csv found
00:00:01 setup: ....  seek for WiFi networks
[  1046][E][WiFiMulti.cpp:55] addAP(): [WIFI][APlistAdd] no ssid or ssid too long

00:00:01 s_ssid=network_01  s_password=COGECO-6D370  s_info=D78C7COGECO    ???????????????????????????????????

[  1062][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: network_01

00:00:01 s_ssid=network_02  s_password=Bouboule  s_info=a4b5c6d7e8                    ??????????????????????????????????

[  1077][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: network_02
[  2697][I][WiFiMulti.cpp:114] run(): [WIFI] scan done
[  2697][I][WiFiMulti.cpp:119] run(): [WIFI] 15 networks found
[  2699][E][WiFiMulti.cpp:191] run(): [WIFI] no matching wifi found!
00:00:02 WiFi credentials are not correct

00:00:03 WiFi credentials are not correct
---- Closed serial port COM3 due to disconnection from the machine ----

schreibfaul1, 2023-03-25:
The order is: SSID{TAB}}Password{TAB}}Info

networks.csv
COGECO-6D370 D78C7COGECO network_01 
Bouboule a4b5c6d7e8 `network_02`

ve2cfb, 2023-03-25:
I am sorry Wolle,
I must be missing something...
Is network_01 suppose to be before the SSID ?

By the way, i got it working by changing the code (dirty way of doing thing) for a test and its working.
I will let you be the judge... 

            for(int i = 0; i < str.length(); i++){
                if(str[i] == '\t'){
 change from 0 to 1                   if(p == 1) s_ssid     = str.substring(q, i);
 change from 1 to 2                    if(p == 2) s_password = str.substring(q, i);
 change from 2 to 3                    if(p == 3) s_info     = str.substring(q, i);
                    p++;
                    i++;
                    q = i;
                }
            }

schreibfaul1, 2023-03-25:
network_01 is just a placeholder for an SSID. 
There will be no SSID with 'network_01' in your home network.

Activate the line
// log_i("s_ssid=%s  s_password=%s  s_info=%s", s_ssid.c_str(), s_password.c_str(), s_info.c_str());
in connectToWiFi()

Then you can see in the log whether the csv is read correctly.

ve2cfb, 2023-03-25:
Ok,
in my last post the log did show everything was offset.
Did remove the 'network_01' and 'network_02'
Everything now works fine.

Thank you many many time.

ve2cfb, 2023-04-01:
Hello Wolle,
Got everything sorted out and i have  3 SSID configured. 
Thanks again.
Pierre

closed 2023-04-20 by ve2cfb