schreibfaul1/ESP32-MiniWebRadio #144: DLNA buttons are not displayed

valera39, 2023-03-25:
Hello Wolle!
There are buttons on the SD card, but they are not displayed on the screen

schreibfaul1, 2023-03-25:
Hello @valera39,
you are ahead of the curve DLNA will soon have its own section, for which I need new buttons.

The error looks like lack of memory. The soap library should be able to process 200 entries per directory. I will test that.

valera39, 2023-03-25:
Thanks for the quick response! I just really like your project, so I'm in such a hurry :)
Is it better to use ESP32S3 C PSRAM?

schreibfaul1, 2023-03-27:
A DLNA directory is no longer transmitted to the web as a complete block, but sequentially each entry one after the other. This takes longer, but no PSRAM is necessary for this and the DLNA directory can be very long.

closed 2023-03-27 by valera39