schreibfaul1/ESP32-MiniWebRadio #145: DLNA

valera39, 2023-03-27:
Hi Wolle!
I sewed the version 2.6.2, but in the list in the browser only the first 100 files are displayed, if you select by files, or 100 folders if you select the sorting by folders.
The following files and folders are not displayed.

schreibfaul1, 2023-03-27:
Yes, there is another blocker that originally exists in the soap lib. You can increase `SOAP_DEFAULT_BROWSE_MAX_COUNT` in SoapESP32.h. I don't know which value makes sense, but a memory overflow should no longer occur.

valera39, 2023-03-27:
I chose SOAP_DEFAULT_BROWSE_MAX_COUNT 300, With large values, reboots.
I will experiment.
I look forward to developing and improving the project
Thank you!

valera39, 2023-03-27:
Now great! I prescribed 500 in SOAP_DEFAULT_BROWSE_MAX_COUNT, works!
Now it would still be a display of the DLNA contents on the display and implement playback by folders or list.
I apologize for inaccuracies, I use Google Translator
Thank you, you are the best!

closed 2023-04-02 by valera39