schreibfaul1/ESP32-MiniWebRadio #146: SD Card Mount Failed

prairietech, 2023-04-02:
I'm getting this error message. Pull-up resistors are in place and I have tried without pull-ups. Using the SD adapter as shown in the additional info folder.  All I have are SanDisk 4gig cards to use. SDMMC freq is at default 20000000.

Noob question, I am assuming the contents of the content_on_SD_Card.zip is simply extracted and copied to the card without the folder name?

Any ideas or do I really need to find other manufactures cards to try?

00:00:00 tft_info: .  init ILI9341
00:00:00 setup: ....  Init SD card
E (198) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107
E (199) vfs_fat_sdmmc: sdmmc_card_init failed (0x107).
[   213][E][SD_MMC.cpp:138] begin(): Failed to initialize the card (0x107). Make sure SD card lines have pull-up resistors 
in place.
E (265) ledc: ledc_get_duty(739): LEDC is not initialized
00:00:00 SD Card Mount Failed

clear-sky, 2023-04-02:
hi
https://github.com/schreibfaul1/ESP32-MiniWebRadio/issues/123

prairietech, 2023-04-02:
Yes I read that post.  My connecting wires are quite short and I have jumpered both of the ground contacts on the adapter.

Now for something different. Power cycled the ESP32 and now this is continuously scrolling in the terminal window:

rst:0x10 (RTCWDT_RTC_RESET),boot:0x37 (SPI_FAST_FLASH_BOOT)
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff

But removing the pull-ups, I'm back to the SD Card Mount Failed issue and the scrolling error above goes away.

schreibfaul1, 2023-04-02:
If the SD card adapter is on a TFT display, series resistors (~100Ohm) could be the problem. These are okay with SPI operation, but cannot be used for SD_MMC.

prairietech, 2023-04-02:
OK, pull-ups removed since I'm running a ILI9341.
Dropped the SDMMC down to 10000000 and still have SD mount failure.
Do you actually recommend trying another brand of card other than Sandisk? I'm using 4gig. Has size been an issue?

clear-sky, 2023-04-05:
i think your problem is socket that you are using
i use a plastic sd sd card adapter and solderd wires directly to the pads
https://github.com/schreibfaul1/ESP32-MiniWebRadio/blob/master/additional_info/SD_Card_Adapter_for_SD_MMC_.jpg
but errors didnot appeared
pulled up all three data lines with 10k resistor and soldered two ground pads of adapter
that solved problem

prairietech, 2023-04-05:
I'm not sure what you mean regarding sd socket?  I did build the same adapter you referenced in the .jpg above and connected both ground pads together.  When I use pull-up resistors, the ESP32 will not boot properly. 
I also tried reducing the SDMMC sample frequency to 10 MHz without success.

clear-sky, 2023-04-06:
sorry for my bad english - yes i mean that plastic adapter
i use esp32-s3 that let me change D0 clk and cmd pins on module 
if you are using esp32 cant change that pins
another problem is length of wires mine is opprox 20cm
my sd card is a chip no name (chinese) 4G  and formatted fat32
copied zip file to sd card and extracted it there
i have heard that someone had success mounting sd card on esp32 with pullup only D0

prairietech, 2023-04-07:
Thanks for the advice!  Maybe I'll try a esp32-s3 instead.
Your information regarding unzipping the file on the SD card is interesting.  
I'll try it this weekend when I have more time.

ve2cfb, 2023-04-11:
Hello,
Let me tell you my esperience with the SD card.
I have tried about every solutions available on this site but nothing worked.

I had a small card reader that has all the data pin with a 10K resistor connected to the +3vcc.
on top of that i change the PINMODE D0 from inputpullup to input only.
Everything works perfect.
Here are a couple picture from : https://oshwlab.com/adrirobot/micro-sd-card-module 

[image]
[image]

Hope this help.

prairietech, 2023-04-11:
More great information, thanks!!  Now with warmer wx, it will take me longer to find time in the shack to build this solution.
73's wb0vhb

closed 2024-02-26 by github-actions[bot]