schreibfaul1/ESP32-MiniWebRadio #147: no sound for aac stations (external dac)

clear-sky, 2023-04-05:
Hi dear wolle
i made this project with esp-s3-12k and pcm5102
mp3 stations are ok but aac not(downloading stream is done but no sound)

schreibfaul1, 2023-04-05:
Hi clear-sky,
maybe the problem is in the PSRAM. With the S3, the PSRAM in the AAC decoder can be used due to the higher data rate (than ESP32). This enables AAC-SBR ('Advanced Audio Coding Spectral Band Replication') to be switched on, which improves the sound. SBR is part of the MPEG4 standard.
The PSRAM is only turned on for the ESP32-S3 if it is defined. (look at aac_decoder.h)
````
#define AAC_ENABLE_MPEG4

#if (defined CONFIG_IDF_TARGET_ESP32S3 && defined BOARD_HAS_PSRAM)
    #define AAC_ENABLE_SBR  // needs additional 60KB DRAM,
#endif
````
You can turn off PSRAM in platform.ini (-DBOARD_HAS_PSRAM)

clear-sky, 2023-04-05:
Hi sir
thanks for reply and thanks for your great work !
i am playing with this project on arduino 
**enabled** psram in qspi mode and problem solved!

closed 2023-04-05 by clear-sky