schreibfaul1/ESP32-MiniWebRadio #149: bug report

clear-sky, 2023-04-06:
Hi all
on weak stations (some times work and some times not)
when it  does not work if you go to sleep timer and exit again it switches to last successful receieved station but with current station (weak) info

clear-sky, 2023-04-06:
in my opinion the best strategy for this stations is reconnecting to url every (for example) 10 seconds!

schreibfaul1, 2023-04-08:
It is currently the case that the last channel from the channel list (no. 001 ... 999) is tuned in after a restart or after the 'time announcement'.
I assume that you have set a station e.g. via the web interface. This has the number 000.
It will be better I connect to the last url after a boot and not the last saved channel number from the list.

clear-sky, 2023-04-08:
Hi thanks for the answer
selecting station via ir remote is awful when stations is more than 10, thus it is better to add selecting them via station number

clear-sky, 2023-04-12:
in web interface UI in logo upload section if logo be smaller than preferred size UI can not stretch it to proper size
in this condition logo contains a withe margin

closed 2023-05-08 by clear-sky