schreibfaul1/ESP32-MiniWebRadio #150: List of stations

valera39, 2023-04-12:
Hello Wolle!
Is it possible to conclude on the display scrolling a list of stations? It seems to me that it would be convenient, scrolling the list with buttons, quickly select the desired station for playback.
And will it be MiniWeb Radio ver.3 ?
Thank you!

clear-sky, 2023-04-12:
Hi all
it needs a drop down menu like tv or stbs
i am not programmer but the easiest solution is using hmi based displays like ft8xx or nextion
waiting for wolle's opinion!

valera39, 2023-04-13:
Why, this list can be compiled on ILI9341 or ILI9488.
As an example
[image]

schreibfaul1, 2023-04-14:
Yes, the layout might be changed in a later version.
At the moment I'm in the process of revising the I2S Lib. Two new decoders have been added, these are the OPUS decoder for ogg (OPUS) streams
http://s1.knixx.fm:5347/dein_webradio_vbr.opus
https://icecast.err.ee/r2rock.opus
and a VORBIS decoder for ogg streams
http://play.global.audio/bgradio_low.ogg
http://play.global.audio/veronika_low.ogg
Both can now be used with MWR, also as podcasts and local files.
What I do now is replace the HELIX AAC decoder with the FDK decoder. So far, it has not been possible for aac+ streams to work in stereo.
When that is done there is no reason to use VS1053 anymore, then I2S is the better choice.

closed 2024-02-26 by github-actions[bot]