schreibfaul1/ESP32-MiniWebRadio #151: editing miniwebradio.csv

clear-sky, 2023-04-14:
Hi all
i have made this radio and trying to explore it
first of all my wifi module is esp-s3-12k this module has 8M flash
i checked partition table and found that it is for 4M modules here is output after compiling the code:

Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  19.4% (used 63524 bytes from 327680 bytes)
Flash: [========  ]  83.0% (used 3153277 bytes from 3801088 bytes)

as you see there is not enough space for additional develop, thus tried to edit it like below:

# Name,     Type,   SubType,   Offset,   Size,     Flags
  phy_init, data,   phy,       0x9000,   0x7000,
  factory,  app,    factory,   0x10000,  0x790000,
  nvs,      data,   nvs,       0x7A0000, 0x40000,

now lets see compiling output:

Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  19.4% (used 63524 bytes from 327680 bytes)
Flash: [====      ]  39.8% (used 3153277 bytes from 7929856 bytes)
Building .pio/build/esp32dev/firmware.bin
esptool.py v4.5
Creating esp32s3 image...
Merged 2 ELF sections
Successfully created esp32s3 image.

but uploading the code fails!!!
please see terminal output:

Serial port /dev/ttyUSB0
Connecting..........

A serial exception error occurred: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
Note: This error originates from pySerial. It is likely not a problem with esptool, but with the hardware connection or drivers.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
*** [upload] Error 1
============================================================================================= [FAILED] Took 31.36 seconds =============================================================================================

 *  The terminal process "platformio 'run', '--target', 'upload'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 

this is my platformio.ini 

; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/en/latest/platforms/espressif32.html

[env:esp32dev]
;platform = https://github.com/platformio/platform-espressif32.git#v5.1.0   ; Arduino V2.0.4
;platform = https://github.com/platformio/platform-espressif32.git#v5.2.0   ; Arduino V2.0.5
platform = https://github.com/platformio/platform-espressif32.git#v6.1.0   ; Arduino V2.0.6
;platform = https://github.com/platformio/platform-espressif32.git

;board = esp32dev ;chipmodel ESP32, 4M FLASH, USBtoTTL
;board = um_tinys3 ;chipmodel ESP32S3, 8M FLASH, HAS_PSRAM, SerialUSB
board = esp32-12k ;chipmodel ESP32S3, 8M FLASH, HAS_PSRAM, USBtoTTL

board_build.f_cpu = 240000000L
board_build.flash_size=8MB
board_build.flash_freq=80M
board_build.spiram_mode=2
framework = arduino
monitor_speed = 115200
monitor_filters = esp32_exception_decoder
board_build.partitions = miniwebradio.csv
upload_speed = 115200  ; 921600, 512000, 460800, 256000, 115200

lib_deps =
    https://github.com/schreibfaul1/ESP32-IR-Remote-Control.git
    https://github.com/schreibfaul1/ESP32-vs1053_ext.git
    https://github.com/schreibfaul1/ESP32-audioI2S.git
    https://github.com/arduino-libraries/Arduino_JSON.git

platform_packages =
;    framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.4
    framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.6
;    framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#master ; newest release

board_upload.maximum_size = 8388608
board_upload.flash_size = 8MB
board_build.flash_mode = qio
board_build.bootloader = dio

;build_flags = -DCORE_DEBUG_LEVEL=0    ; None
;build_flags = -DCORE_DEBUG_LEVEL=1    ; Error
;build_flags = -DCORE_DEBUG_LEVEL=2    ; Warn
;build_flags = -DCORE_DEBUG_LEVEL=3    ; Info
;build_flags = -DCORE_DEBUG_LEVEL=4    ; Debug
;build_flags = -DCORE_DEBUG_LEVEL=5    ; Verbose

build_flags =
        -Wall
        -Wextra
        -std=c++17
        -std=gnu++17
        -Wdouble-promotion             ; double to float warnings
        -Wimplicit-fallthrough         ; switch case without break
        -DCORE_DEBUG_LEVEL=3
        -DCONFIG_ARDUHAL_LOG_COLORS
        -DBOARD_HAS_PSRAM
        -DARDUINO_RUNNING_CORE=1       ; Arduino Runs On Core (setup, loop)
        -DARDUINO_EVENT_RUNNING_CORE=1 ; Events Run On Core
;        -DAUDIO_LOG

build_unflags =
;       -DARDUINO_USB_CDC_ON_BOOT=0    ; traditional log
;       -DBOARD_HAS_PSRAM

after this Disgraceful failure tried this trick in arduino:
copied .scv file to this path:

/home/conrad/.arduino15/packages/esp32/hardware/esp32/2.0.7/tools/partitions/

and edited boards.txt in this path:

/home/conrad/.arduino15/packages/esp32/hardware/esp32/2.0.7/

add this lines to partitions names section:

esp32s3.menu.PartitionScheme.miniwebradio=MiniWebRadio (Ultra Huge App 7.5625MB + 250KB NVS)
esp32s3.menu.PartitionScheme.miniwebradio.build.partitions=miniwebradio
esp32s3.menu.PartitionScheme.miniwebradio.upload.maximum_size=7929856

wooow!!! it works in arduino 1.8.19
uploading without problem

but in platformio what is wrong?

schreibfaul1, 2023-04-14:
It's not a memory problem. There are different upload variants, I think you need native USB and not a UART simulation.
Look at https://github.com/platformio/platform-espressif32/tree/develop/boards
Try another board in platform.io, probably esp32-s3-devkitc-1

clear-sky, 2023-04-15:
Hi Dear Wolle
thanks for the answer
my problem solved 
just installed platformio udev rules
https://docs.platformio.org/en/stable/core/installation/udev-rules.html

closed 2023-04-15 by clear-sky

clear-sky, 2023-04-20:
for linux users only

for people that want what is the problem exactly
my os is linux and i did not have any axperience on platformio
in first step installed code ( free version of vscode)and tried to install platformio but did not worked
in second try installed code marketplace to enable finding platformio
platformio found and installed but it forced me to install microsoft standard c++ library 
that was very odd and made me toooooo angry! because linux has the best c++ compiler
on the other hand i tried to continue testing vscode
on first step compiled and uploaded the mini web radio and worked perfectly
then tried to play with it 
colsed the program and edited miniwebradio.csv file then reopend code (vscode)and tried to uploaded program
i get this error when uploading!!!
this is a bug in vscode - let me tell what happend exactly
at first upload test after uploading opened serial terminal monitor and closed code (vscode)
when you close vscode the program does not release /dev/ttyUSB0 without any message
when i try to reopen vscode when uploading program gave that error message 
in fact installing udev-rules did not solve the problem because after installing i rebooted system and /dev/ttyUSB0 released
if anyone wants to solve this problem must to close serial monitor before closing vscode