schreibfaul1/ESP32-MiniWebRadio #152: Music terminates prematurely when .mp3 and .m4a files in same playlist.

rjshiffer, 2023-04-15:
Firstly, the radio is fantastic and my wife and I play it daily.  At her request
I uploaded her favorite songs to the SD and created customized playlists.  I 
noticed that after a few songs played the remaining songs in the playlist ended
early.  In fact, each of the remaining songs stopped after exactly 39 seconds.
After much troubleshooting the only pattern I noticed was the problem occurred
when .m4a files played after .mp3 files.

I made a small, test playlist to duplicate the problem.  
 
```
list1.m3u

#EXTM3U
#EXTINF:164,Four Tops - I Can't Help Myself
CD/Motown/D1/1-12 I Can't Help Myself.m4a
#EXTINF:164,Four Tops - I Can't Help Myself
CD/Motown/D1/1-12 I Can't Help Myself.m4a
#EXTINF:185,I Could Never Love Another
Songs/I Could Never Love Another (After.mp3
#EXTINF:164,Four Tops - I Can't Help Myself
CD/Motown/D1/1-12 I Can't Help Myself.m4a
#EXTINF:164,Four Tops - I Can't Help Myself
CD/Motown/D1/1-12 I Can't Help Myself.m4a
```

The first two .m4a songs play perfectly followed by the .mp3 song.  The last 
2 songs (the same as the first 2) each stop exactly after 39 seconds.  If I
play other .m4a songs from another playlist they all end after 39 seconds until
I boot.  After I boot, I could play all these (plus more) .m4a songs successfully
until the next .mp3 song is played.

Any ideas?  I going crazy trying to figure this out.

Thanks,
Richard

```

Here is the log of list1.m3u playing:

00:00:00    
00:00:00        ***************************    
00:00:00        *     MiniWebRadio V2     *    
00:00:00        ***************************    
00:00:00    
00:00:00 setup: ....  Arduino is pinned to core 1
00:00:00 tft_info: .  init ILI9341
00:00:00 setup: ....  Init SD card
00:00:00 setup: ....  SD card found
E (835) ledc: ledc_get_duty(739): LEDC is not initialized
00:00:01 setup: ....  seek for stations.csv
00:00:01 setup: ....  stations.csv found
00:00:01 setup: ....  seek for WiFi networks
00:00:05 setup: ....  connected to XXXXX, IP address is XXXXXXXXXX
00:00:05 ftpServer:   Buffers allocated: 4622 bytes
00:00:05 rtime_info:  Initializing SNTP
00:00:05 rtime_info:  Waiting for system time to be set... (1/10)
18:56:19 rtime_info:  18:56:19
18:56:19 audiotask:   is pinned to core 1
18:56:19 audiotask:   priority is 2
18:56:19 setup: ....  Number of saved stations: 19
18:56:19 setup: ....  current station number: 1
18:56:19 setup: ....  current volume: 11
18:56:19 VS1053 chipID = 1844972805, version = 3
18:56:19 action: ...  current volume is 11
18:56:19 action: ...  switch to station 1
18:56:19 AUDIO_info:  Song stopped correctly after 0 msec
18:56:19 AUDIO_info:  PSRAM found, inputBufferSize: 295903 bytes
18:56:19 AUDIO_info:  Connect to new host: "edge-bauerall-01-gos2.sharp-stream.com:80/net2lincoln.mp3?aw_0_1st.skey=1653967769&aw_0_1st.playerid=BMUK_RPi"
18:56:19 AUDIO_info:  Connect to "edge-bauerall-01-gos2.sharp-stream.com" on port 80, extension "/net2lincoln.mp3?aw_0_1st.skey=1653967769&aw_0_1st.playerid=BMUK_RPi"
18:56:19 AUDIO_info:  Connected to server in 162 ms
18:56:20 WiFI_info:   RSSI is -47 dB
18:56:20 DLNA_info:   SSDP search for media servers started
18:56:20 AUDIO_info:  HTTP/1.0 200 OK
18:56:20 AUDIO_info:  content-type: audio/mpeg, format is mp3
18:56:20 AUDIO_info:  date: Fri, 14 Apr 2023 23:56:20 GMT
18:56:20 bitRate:     0Kbit/s
18:56:20 AUDIO_info:  icy-name: Greatest Hits Radio (Lincolnshire)
18:56:20 StationName: Greatest Hits Radio (Lincolnshire)
18:56:20 AUDIO_info:  server: Icecast 2.4.0-kh15
18:56:20 AUDIO_info:  access-control-allow-origin: *
18:56:20 AUDIO_info:  access-control-allow-headers: Origin, Accept, X-Requested-With, Content-Type, Icy-MetaData
18:56:20 AUDIO_info:  access-control-allow-methods: GET, OPTIONS, SOURCE, PUT, HEAD, STATS
18:56:20 AUDIO_info:  Switch to DATA, metaint is 16000
18:56:20 lastURL: ..  edge-bauerall-01-gos2.sharp-stream.com:80/net2lincoln.mp3?aw_0_1st.skey=1653967769&aw_0_1st.playerid=BMUK_RPi
18:56:20 AUDIO_info:  stream read
18:56:20 AUDIO_info:  buffer filled in 3 ms
18:56:20 AUDIO_info:  StreamTitle='You Gotta Be - Des'Ree'
18:56:20 StreamTitle: You Gotta Be - Des'Re
18:56:20 AUDIO_info:  StreamUrl='https://listenapi.planetradio.co.uk/api9.2/eventdata/181174445'
18:56:28 DLNA_info:   SSDP query discovered 1 media servers
18:56:28 DLNA_info:   connected successfully to XXXXXXXXX:XXXX
18:56:28 ftpServer:   Ftp server waiting for connection on port 21

                      list1.m3u starts playing here

18:57:05 Playlist:    #EXTM3U

18:57:05 Playlist:    #EXTINF:164,Four Tops - I Can't Help Myself

18:57:05 Playlist:    CD/Motown/D1/1-12 I Can't Help Myself.m4a
18:57:05 AUDIO_info:  Song stopped correctly after 0 msec
18:57:05 AUDIO_info:  Reading file: "/audiofiles/CD/Motown/D1/1-12 I Can't Help Myself.m4a"
18:57:05 AUDIO_info:  stream ready
18:59:49 AUDIO_info:  Song stopped correctly after 0 msec
18:59:49 AUDIO_info:  End of file "1-12 I Can't Help Myself.m4a"
18:59:49 end of file: 1-12 I Can't Help Myself.m4a
18:59:49 Playlist:    #EXTINF:164,Four Tops - I Can't Help Myself

18:59:49 Playlist:    CD/Motown/D1/1-12 I Can't Help Myself.m4a
18:59:49 AUDIO_info:  Song stopped correctly after 0 msec
18:59:49 AUDIO_info:  Reading file: "/audiofiles/CD/Motown/D1/1-12 I Can't Help Myself.m4a"
18:59:50 AUDIO_info:  stream ready
19:02:33 AUDIO_info:  Song stopped correctly after 0 msec
19:02:33 AUDIO_info:  End of file "1-12 I Can't Help Myself.m4a"
19:02:33 end of file: 1-12 I Can't Help Myself.m4a
19:02:33 Playlist:    #EXTINF:185,I Could Never Love Another

19:02:33 Playlist:    Songs/I Could Never Love Another (After.mp3
19:02:34 AUDIO_info:  Song stopped correctly after 0 msec
19:02:34 AUDIO_info:  Reading file: "/audiofiles/Songs/I Could Never Love Another (After.mp3"
19:02:34 AUDIO_info:  Content-Length: 1514645
19:02:34 AUDIO_info:  ID3 framesSize: 28672
19:02:34 AUDIO_info:  ID3 version: 2.3
19:02:34 AUDIO_info:  ID3 normal frames
19:02:34 id3data: ..  Title: I Could Never Love Another (After Loving You)
19:02:34 id3data: ..  Track: 7
19:02:34 id3data: ..  Artist: The Temptations
19:02:34 id3data: ..  ContentType: Rock/Pop
19:02:34 id3data: ..  Album: Live at the Copa/With a Lot of Soul
19:02:34 id3data: ..  Length (ms): 185730
19:02:34 AUDIO_info:  Audio-Length: 1485973
19:02:34 AUDIO_info:  stream ready
19:04:27 WiFI_info:   RSSI is -67 dB
19:04:28 WiFI_info:   RSSI is -47 dB
19:04:32 WiFI_info:   RSSI is -51 dB
19:04:42 WiFI_info:   RSSI is -48 dB
19:05:22 WiFI_info:   RSSI is -51 dB
19:05:38 WiFI_info:   RSSI is -55 dB
19:05:39 WiFI_info:   RSSI is -49 dB
19:05:39 AUDIO_info:  Song stopped correctly after 20 msec
19:05:39 AUDIO_info:  End of file "I Could Never Love Another (After.mp3"
19:05:39 end of file: I Could Never Love Another (After.mp3
19:05:39 Playlist:    #EXTINF:164,Four Tops - I Can't Help Myself

19:05:39 Playlist:    CD/Motown/D1/1-12 I Can't Help Myself.m4a
19:05:40 AUDIO_info:  Song stopped correctly after 0 msec
19:05:40 AUDIO_info:  Reading file: "/audiofiles/CD/Motown/D1/1-12 I Can't Help Myself.m4a"
19:05:40 AUDIO_info:  stream ready
19:06:19 AUDIO_info:  Song stopped correctly after 0 msec
19:06:19 AUDIO_info:  End of file "1-12 I Can't Help Myself.m4a"
19:06:19 end of file: 1-12 I Can't Help Myself.m4a
19:06:19 Playlist:    #EXTINF:164,Four Tops - I Can't Help Myself

19:06:19 Playlist:    CD/Motown/D1/1-12 I Can't Help Myself.m4a
19:06:20 AUDIO_info:  Song stopped correctly after 0 msec
19:06:20 AUDIO_info:  Reading file: "/audiofiles/CD/Motown/D1/1-12 I Can't Help Myself.m4a"
19:06:20 AUDIO_info:  stream read
19:06:59 AUDIO_info:  Song stopped correctly after 0 msec
19:06:59 AUDIO_info:  End of file "1-12 I Can't Help Myself.m4a"
19:06:59 end of file: 1-12 I Can't Help Myself.m4a
19:06:59 end of playlist

Song  Fmt   Start      End        Elasped  Success?
  1   .m4a  18:57:05   18:59:49   2m 44s    OK
  2   .m4a  18:59:49   19:02:33   2m 44s    OK 
  3   .mp3  19:02:34   19:05:39   3m  5s    OK
  4   .m4a  19:05:40   19:06:19      39s    Fail
  5   .m4a  19:06:20   19:06:59      39s    Fail 

```

Additional information:  If I replace the .mp3 song with a different one,  the shorter durations of the .m4a song changes.  For example,  the following .m4a songs may all play for 44 seconds or all play for 50 seconds depending on the .mp3 song.

schreibfaul1, 2023-04-16:
Hello @rjshiffer,
unbelievable, but I can understand the problem. Instead of the playlist with many short files, I created one with several long songs with different formats. I think this is independent of the format used m4a or mp3 doesn't seem to matter.
Thanks for the report, I don't know the cause of this 'permanent loop' yet, but I'll go troubleshooting in the coming week.
Until then, many greetings from Berlin
Wolle

schreibfaul1, 2023-04-18:
Hope the bug is fixed now. I couldn't pinpoint the cause, it's probably a timing issue. I have a 5s timeout in case the destination or specified file cannot be played.
The playlist I created for this with a running time of about 2 hours ran without any problems.

rjshiffer, 2023-04-19:
Hello Wolle,

Unfortunately the fix did not resolve my problem.  I ran a few other combinations that may be helpful.

1.  I loaded an .mp3 file and an .m4a file directly into the /audiofiles directory.  Boot, play .mp3 immediately, followed by the .m4a.  The .m4a files ends early.  I play the .mp3 file again and it completes normally.  That is, the problem is independent of playlists.

2. Boot.  Play the .mp3 file from the DLNA server, followed by the .m4a file.  Both files play normally.

I only have .mp3 and .m4a files so cannot play other formats.  I didn't collect log files but can do if that would be helpful.

Again, thanks from Chicago, Illinois, USA

Richard

rjshiffer, 2023-04-19:
Hi  Wolle,

As only 10-12 of the 120 songs were .mp3 format, I converted the .mp3 files to .m4a format.  All is now good.  I don't know what is unique about the .mp3 files causing the problem but this work around solves it at least for now.   

Thanks for your efforts,
Richard

schreibfaul1, 2023-04-19:
Hi Richard,
very nice if it works now. Thanks to your hint, I was able to find an error in the processing of the playlist. If the URL is incorrect or the local audio file does not exist, the next entry should be executed after 5s. For this I have provided a time limit that was not effective.
In your log I saw that you are using a VS1053. My VS1053 boards only work well with aac and mp3, m4a and flac is poor. An external DAC like PCM5102 is more reliable.

best regards
Wolle

closed 2023-04-22 by rjshiffer