schreibfaul1/ESP32-MiniWebRadio #153: audio bitrate

clear-sky, 2023-04-21:
hi dear wolle
in normal stations audio bit rate is shown in serial terminal in two sections:
1-bitRate:
2-AUDIO_info:
in some stations program can not detect audio bit rate 
in this stations i can not see bitRate line 
for example in this station
https://radio.iraninternational.app/iintl_c

please check!

schreibfaul1, 2023-04-21:
Hi @clear-sky,
yes, at the moment it is like this, the bitrate is read from the HTTP response header. If the radio station does not transmit a bitrate, nothing is displayed.
I could also read the bitrate from the decoders (the audioI2S Lib), with a variable bitrate you then have to wait a few samples until the bitrate is determined.
With the VS1053 I haven't planned that at all. I believe there is a WRAM address in the VS1053 from which the bitrate can be read. In the coming days I will see if I would work this out.

clear-sky, 2023-05-03:
Hi dear wolle !
in new branch  i dont see this problem (run-Arduino-as--a-component)

schreibfaul1, 2023-05-03:
That would be fine, the branch will later become the master. I'm still trying to optimize the branch.
It is so that when the library detects that the target is no longer responding, the connection is re-established. This is the case, for example, if the provider (after 24 hours) interrupts the existing Internet connection and your router reconnects. Then the assigned IP address changes. And this is then communicated to the destination so that 24/7 operation is possible.

But I don't know exactly what changed. I couldn't take over the Arduino settings completely, it just didn't work.
If you want to change or optimize something yourself, then open the terminal in VSCode and enter: `pio run -t menuconfig`

clear-sky, 2023-05-03:
Hi dear wolle!
thanks-very good news
will try it
*thanks with respect*

closed 2023-05-08 by clear-sky