schreibfaul1/ESP32-MiniWebRadio #156: MP3 decode error -2 : MAINDATA_UNDERFLOW

clear-sky, 2023-04-28:
Hi dear wolle
please try this stream 
http://us3.internet-radio.com:8342/stream
in my pc is ok but in radio i alwas get this errors: 
```
21:06:16 AUDIO_info:  icy-name: TOPBLUES
21:06:16 StationName: TOPBLUES
21:06:16 bitRate:     320Kbit/s
21:06:17 icy-url: ..  http://www.topblues.com
21:06:17 AUDIO_info:  MP3Decoder has been initialized, free Heap: 182624 bytes , free stack 5460 DWORDs
21:06:17 lastURL: ..  http://us3.internet-radio.com:8342/stream
21:06:17 AUDIO_info:  stream ready
21:06:17 AUDIO_info:  syncword found at pos 0
21:06:17 AUDIO_info:  MP3 decode error -2 : MAINDATA_UNDERFLOW
21:06:17 AUDIO_info:  syncword found at pos 1043
21:06:17 AUDIO_info:  syncword found at pos 0
21:06:17 AUDIO_info:  Channels: 2
21:06:17 AUDIO_info:  SampleRate: 44100
21:06:17 AUDIO_info:  BitsPerSample: 16
21:06:17 AUDIO_info:  BitRate: 320000
21:06:17 AUDIO_info:  StreamTitle='Ronnie Earl - Navajo Blues [4pQJ]'
21:06:17 StreamTitle: Ronnie Earl - Navajo Blues [4pQJ]
21:06:18 AUDIO_info:  slow stream, dropouts are possible
21:06:19 AUDIO_info:  slow stream, dropouts are possible
21:06:20 AUDIO_info:  slow stream, dropouts are possible
21:06:21 AUDIO_info:  slow stream, dropouts are possible
21:06:22 AUDIO_info:  slow stream, dropouts are possible
```

schreibfaul1, 2023-05-02:
Difficult thing, http://us3.internet-radio.com:8342/stream transmits many small data packets at 320Kbit/s. That's not to do. I suspected WiFiClient at first, but that's not the case because it's no better in ESP-IDF without WiFiClient either.
The solution I see is to adjust the TCP settings. But that's not easy with a precompiled Arduino.

I created a branch `run Arduino as a component`. To select this, copy the URL and clone the repository as usual. Then you go to the task bar on the bottom left and click on 'master', then you can select the new branch. The IDE is fetching the framework and libraries, please wait until the update is successful. Then you can adjust the HW in common.h, compile and upload then. It feels like this is taking a lot longer than usual. After that, TOPBLUES plays without any problems and the audio buffer is well filled.

clear-sky, 2023-05-03:
thank you very much for your hard work
as far as i understood you add idf and cmake
problem solved ! yay!!!! 

 i am new to programming but can understand you are making a reliable foundation for web radio that nobody made it before!
last year i made a webradio based on edzelf repo it was based on vs1053 but had serious problems with 320k and https streams -it disappointed me
but this year switched to your repo because i saw your great work for creating software decoding audio codecs with external dac

*thanks again with respect*

closed 2023-05-03 by clear-sky