schreibfaul1/ESP32-MiniWebRadio #160: module lifetime?

clear-sky, 2023-05-08:
Hi all
as far as i understood some data such as stations volume and bass -treble  etc will be read from sd card and will be written in nvs
when user listen to radio and for example changes station new info such as new volume capacity and last stations will be rewrite to nvs
my question is every change written in nvs is equall to whole flashing module?
if answer is yes the life of flash will be ended soon?(flash life time is between 10000 to 100000 cycles)

schreibfaul1, 2023-05-11:
Yes, that's right, the last station number and current volume is stored in NVS. It is possible that after a long time a memory cell will fail, but that is unlikely to happen.

clear-sky, 2023-05-11:
hi dear wolle
thanks for the answer-is there any way to store this data in some where out of module? like sd card?

schreibfaul1, 2023-05-11:
I could do that, maybe a settings.ini file on SD_MMC. The question is whether it is worth the effort.

clear-sky, 2023-05-11:
i have a nice rtc and love it very much!!!
ds3234 has spi interface and has 254 bytes sram
is that space enough for that purpose?

clear-sky, 2023-05-11:
It's just between you and me!!! At first, I thought that the settings are saved in sd card

clear-sky, 2023-05-13:
`The question is whether it is worth the effort`
Hello Dear Wolle!
it depends on you-if you want to make a reliable radio the answer is yes

schreibfaul1, 2023-05-13:
eallo clear-sky,
using an RTC ds3234 would only outsource the problem.
I think it's not really a problem, the memory cells of the EEPROM will surely survive the whole lifetime of an ESP32. I have already sorted out ESPs with defective PSRAM or other errors. I'll think about storing it on the SD card.

clear-sky, 2023-05-14:
Hi Dear Wolle!
Thanks for the answer
in my opinion using sram is a great idea but the problem is ds3234 is rare
you are right ! using sd card is the best choise With the current situation
on the other hand i am testing your radio HARD! if a problem happend will tell you
*tanks with respect*

closed 2023-05-14 by clear-sky

clear-sky, 2023-05-18:
wooow!!!!
thanks for your great work!
by the way what is audio protocol logos in common folder?(sd card)
sorry because my display does not have touch!
*thanks with respect*