schreibfaul1/ESP32-MiniWebRadio #162: The touchscreen stopped working

valera39, 2023-05-10:
Hello Wolle!
I apologize for the Google translation.
I always look forward to your updates.
I moved to version 2.7.1, and my touch screen stopped working. If you carefully look at the backlight, then the moment of pressing the screen, the backlight barely blinks, but the buttons are not displayed on the screen.
I did not change the compounds. He returned to version 2.6.3 - works.
Configuration: ESP32 S3, VS1053, ILI9488.
Thank you for updating the project, I hope my wishes [in](https://github.com/schreibfaul1/ESP32-MiniWebRadio/issues/150) will be implemented in the future.

schreibfaul1, 2023-05-10:
Hello Valera,
with the switch from the precompiled Arduino to the IDF, there are now different environments for ESP32 and ESP32-S3. 
Did you choose the right environment?
[image]

valera39, 2023-05-10:
Yes, I did so

valera39, 2023-05-10:
At the beginning of the assembly there is a phrase: Warning! Arduino framework as an ESP-IDF component doesn't handle the `variant` field! The default `esp32` variant will be used.
Could it be a problem?

schreibfaul1, 2023-05-10:
I have tested the ESP32-S3, vs1053, ILI9488 configuration. I can not reproduce the error.

`Warning! Arduino framework as an ESP-IDF component doesn't handle the `variant` field! The default `esp32` variant will be used. You can ignore it.

I suspect a faulty build. Maybe a clean will help?
[image]

valera39, 2023-05-10:
I think that the connection is reliable because version 2.6.1 works excellently
I reinstalled VS Code and Platformio and sewed it again. And finally, the touchscreen earned, but strange.
I shot a small video with sound, maybe you will understand [more.](https://files.fm/f/fscjz9the)
When pressing the Mute button, a squeak appears in the dynamics and with a repeated pressing noise
Instead of sound. Also pay attention to displaying stations switching buttons. Power of the radio from the pauerbank.
And once again, in version 2.6.1 there are no such problems, everything is excellent switches and there is absolutely no noise.
Tomorrow I'll try to replace ESP32 S3 with ESP32-12K.
Thank you!

valera39, 2023-05-10:
I think that the connection is reliable because version 2.6.1 works excellently
I reinstalled VS Code and Platformio and sewed it again. And finally, the touchscreen earned, but strange.
I shot a small video with sound, maybe you will understand [more.](https://files.fm/f/fscjz9the)
When pressing the Mute button, a squeak appears in the dynamics and with a repeated pressing noise
Instead of sound. Also pay attention to displaying stations switching buttons. Power of the radio from the pauerbank.
And once again, in version 2.6.1 there are no such problems, everything is excellent switches and there is absolutely no noise.
Tomorrow I'll try to replace ESP32 S3 with ESP32S3-12K.
Thank you!

schreibfaul1, 2023-05-11:
I think I know what you mean now. That's how it should be.

https://github.com/schreibfaul1/ESP32-MiniWebRadio/assets/26044260/209c57af-6321-46ed-9225-499b9456b1ef

SD_MMC has a bug that reduces the speed, this should be fixed with the next big Arduino version (V3.0.0) change. Then the graphics are smoother.

I'm in the process of replacing the AAC decoder in the audioI2S so that aacp and aac+ can be played back in stereo. This is the case with almost all M3U8 streams. The VS1053 can already do this.

schreibfaul1, 2023-05-11:
It looks like your ESP32-S3 doesn't have PSRAM. I had increased the audio buffer, but this may cause memory problems. I reset the audio buffer to its original size.

valera39, 2023-05-11:
I replaced ESP32S3 from Espressif with ESP32S3-12K from AI-thinker. I think this is the right choice. In firmware 2.7, it is impossible to avoid extraneous noise at zero volume. Maybe this can be done by changing the code? 
And can I make a "stop" button instead of "mute" to turn off the stream?
Thanks!

closed 2023-05-18 by valera39