schreibfaul1/ESP32-MiniWebRadio #164: no Audio in V2.7

Arne65, 2023-05-15:
Hallo Wolle
A great project, thank you very much

in V2.7 I have neither with VS1053 nor with PCM5102A audio.
In V2.6 both boards run without problems. Can you maybe give me a tip?
Best regards

schreibfaul1, 2023-05-15:
Hello Arne65,
another user has already reported a similar problem. It looks as if the switching of the environments (ESP32, ESP32-S3) no longer works as safely as before. 
I have therefore adapted the platformio.ini, which now switches both targets separately. Hope that's the cause, I'm not sure, but I'd be happy if you test this again.

Arne65, 2023-05-16:
Hello Wolle

Thanks for your quick reaction. I tested it, but unfortunately it doesn't work. When compiling, I noticed that the following warnings appeared.

Compiling .pio\build\esp32\Preferences.o
C:/Users/jop/.platformio/packages/framework-arduinoespressif32-src-52b8cdec3e60619426e51a1d18d1c3fa/libraries/I2S/src/I2S.cpp: In member function 'int I2SClass::_installDriver()':
C:/Users/jop/.platformio/packages/framework-arduinoespressif32-src-52b8cdec3e60619426e51a1d18d1c3fa/libraries/I2S/src/I2S.cpp:148:3: warning: missing initializer for member 'esp_i2s::i2s_driver_config_t::tx_desc_auto_clear' [-Wmissing-field-initializers]
   };
   ^
C:/Users/jop/.platformio/packages/framework-arduinoespressif32-src-52b8cdec3e60619426e51a1d18d1c3fa/libraries/I2S/src/I2S.cpp:148:3: warning: missing initializer for member 'esp_i2s::i2s_driver_config_t::fixed_mclk' [-Wmissing-field-initializers]
C:/Users/jop/.platformio/packages/framework-arduinoespressif32-src-52b8cdec3e60619426e51a1d18d1c3fa/libraries/I2S/src/I2S.cpp:148:3: warning: missing initializer for member 'esp_i2s::i2s_driver_config_t::mclk_multiple' [-Wmissing-field-initializers]
C:/Users/jop/.platformio/packages/framework-arduinoespressif32-src-52b8cdec3e60619426e51a1d18d1c3fa/libraries/I2S/src/I2S.cpp:148:3: warning: missing initializer for member 'esp_i2s::i2s_driver_config_t::bits_per_chan' [-Wmissing-field-initializers]
C:/Users/jop/.platformio/packages/framework-arduinoespressif32-src-52b8cdec3e60619426e51a1d18d1c3fa/libraries/I2S/src/I2S.cpp: In member function 'int I2SClass::_applyPinSetting()':
C:/Users/jop/.platformio/packages/framework-arduinoespressif32-src-52b8cdec3e60619426e51a1d18d1c3fa/libraries/I2S/src/I2S.cpp:324:5: warning: missing initializer for member 'esp_i2s::i2s_pin_config_t::mck_io_num' [-Wmissing-field-initializers]
     };
     ^
In file included from C:/Users/jop/.platformio/packages/framework-arduinoespressif32-src-52b8cdec3e60619426e51a1d18d1c3fa/cores/esp32/esp32-hal.h:80,
                 from C:/Users/jop/.platformio/packages/framework-arduinoespressif32-src-52b8cdec3e60619426e51a1d18d1c3fa/cores/esp32/Arduino.h:36,
                 from C:/Users/jop/.platformio/packages/framework-arduinoespressif32-src-52b8cdec3e60619426e51a1d18d1c3fa/libraries/I2S/src/I2S.cpp:19:
C:/Users/jop/.platformio/packages/framework-arduinoespressif32-src-52b8cdec3e60619426e51a1d18d1c3fa/libraries/I2S/src/I2S.cpp: In member function 'int I2SClass::_gpioToAdcUnit(gpio_num_t, esp_i2s::adc_unit_t*)':
C:/Users/jop/.platformio/packages/framework-arduinoespressif32-src-52b8cdec3e60619426e51a1d18d1c3fa/cores/esp32/esp32-hal-log.h:143:38: warning: this statement may fall through [-Wimplicit-fallthrough=] #define log_w(format, ...) log_printf(ARDUHAL_LOG_FORMAT(W, format), ##__VA_ARGS__)
                            ~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:/Users/jop/.platformio/packages/framework-arduinoespressif32-src-52b8cdec3e60619426e51a1d18d1c3fa/libraries/I2S/src/I2S.cpp:1068:7: note: in expansion of macro 'log_w'
       log_w("GPIO 0 for ADC should not be used for dev boards due to external auto program circuits.");
       ^~~~~
C:/Users/jop/.platformio/packages/framework-arduinoespressif32-src-52b8cdec3e60619426e51a1d18d1c3fa/libraries/I2S/src/I2S.cpp:1069:5: note: here
     case GPIO_NUM_4:

Compiling .pio\build\esp32s3\Preferences.o
C:/Users/jop/.platformio/packages/framework-arduinoespressif32-src-52b8cdec3e60619426e51a1d18d1c3fa/libraries/I2S/src/I2S.cpp: In member function 'int I2SClass::_installDriver()':
C:/Users/jop/.platformio/packages/framework-arduinoespressif32-src-52b8cdec3e60619426e51a1d18d1c3fa/libraries/I2S/src/I2S.cpp:148:3: warning: missing initializer for member 'esp_i2s::i2s_driver_config_t::tx_desc_auto_clear' [-Wmissing-field-initializers]
   };
   ^
C:/Users/jop/.platformio/packages/framework-arduinoespressif32-src-52b8cdec3e60619426e51a1d18d1c3fa/libraries/I2S/src/I2S.cpp:148:3: warning: missing initializer for member 'esp_i2s::i2s_driver_config_t::fixed_mclk' [-Wmissing-field-initializers]
C:/Users/jop/.platformio/packages/framework-arduinoespressif32-src-52b8cdec3e60619426e51a1d18d1c3fa/libraries/I2S/src/I2S.cpp:148:3: warning: missing initializer for member 'esp_i2s::i2s_driver_config_t::mclk_multiple' [-Wmissing-field-initializers]
C:/Users/jop/.platformio/packages/framework-arduinoespressif32-src-52b8cdec3e60619426e51a1d18d1c3fa/libraries/I2S/src/I2S.cpp:148:3: warning: missing initializer for member 'esp_i2s::i2s_driver_config_t::bits_per_chan' [-Wmissing-field-initializers]
C:/Users/jop/.platformio/packages/framework-arduinoespressif32-src-52b8cdec3e60619426e51a1d18d1c3fa/libraries/I2S/src/I2S.cpp:148:3: warning: missing initializer for member 'esp_i2s::i2s_driver_config_t::chan_mask' [-Wmissing-field-initializers]
C:/Users/jop/.platformio/packages/framework-arduinoespressif32-src-52b8cdec3e60619426e51a1d18d1c3fa/libraries/I2S/src/I2S.cpp:148:3: warning: missing initializer for member 'esp_i2s::i2s_driver_config_t::total_chan' [-Wmissing-field-initializers]
C:/Users/jop/.platformio/packages/framework-arduinoespressif32-src-52b8cdec3e60619426e51a1d18d1c3fa/libraries/I2S/src/I2S.cpp:148:3: warning: missing initializer for member 'esp_i2s::i2s_driver_config_t::left_align' [-Wmissing-field-initializers]
C:/Users/jop/.platformio/packages/framework-arduinoespressif32-src-52b8cdec3e60619426e51a1d18d1c3fa/libraries/I2S/src/I2S.cpp:148:3: warning: missing initializer for member 'esp_i2s::i2s_driver_config_ [... cut ...]
C:/Users/jop/.platformio/packages/framework-arduinoespressif32-src-52b8cdec3e60619426e51a1d18d1c3fa/libraries/I2S/src/I2S.cpp:148:3: warning: missing initializer for member 'esp_i2s::i2s_driver_config_t::skip_msk' [-Wmissing-field-initializers]
C:/Users/jop/.platformio/packages/framework-arduinoespressif32-src-52b8cdec3e60619426e51a1d18d1c3fa/libraries/I2S/src/I2S.cpp: In member function 'int I2SClass::_applyPinSetting()':
C:/Users/jop/.platformio/packages/framework-arduinoespressif32-src-52b8cdec3e60619426e51a1d18d1c3fa/libraries/I2S/src/I2S.cpp:324:5: warning: missing initializer for member 'esp_i2s::i2s_pin_config_t::mck_io_num' [-Wmissing-field-initializers]

schreibfaul1, 2023-05-17:
The warnings `[-Wmissing-field-initializers]` and `[-Wimplicit-fallthrough=]` are harmless. These come from the Arduino area, we can't do anything about them. Maybe your compiler considers warnings as errors. You can suppress the messages, set the compiler flags `-Wall` and `-Wextra` from `build flags` to `build unflags` in platformio.ini..
[image]

PS You don't need to compile both environments, esp32 or esp32-s3 is enough. To do this, set the field 'Default (ESP32-MiniWebRadio)' to your environment in the taskbar below.
[image]

Arne65, 2023-05-17:
Thanks for the feedback. I'll test it again with an ESP32-S3, maybe the sound will work. And then I give feedback

ivp1967, 2023-05-17:
Good afternoon. I also have no sound when I install version 2.7 ESP32 and VS1053.

schreibfaul1, 2023-05-17:
If there is no PSRAM, this must now be announced. This is no longer recognized automatically, no idea why? With the precompiled Arduino, this always worked reliably. If so, please delete `-DBOARD_HAS_PSRAM` in platformio.ini.

Arne65, 2023-05-17:
Hello Wolle

now works with ESP 32 thanks for the quick help
ESP32-S3-DevKitC-1-N8R2 is also running
THANKS

schreibfaul1, 2023-05-17:
Thank you for your feedback. I had the expectation that the switch from the precompiled Arduino would be easier. Unfortunately, I couldn't adopt the sdkconfig 1:1, and the current state won't be the best. If you still have ideas, you can click the ant in VSCode and then call Platform/Run Menuconfig to change settings yourself.

best regards from Berlin
Wolle

closed 2023-06-26 by Arne65