schreibfaul1/ESP32-MiniWebRadio #165: Wiring Question

BruceKenley, 2023-05-17:
hello, I need some advice on the VS1053 wiring. Being connected to HSPI l thought, the wiring should be 13,mosi,12,miso,14,clk.15cs. The circuit shows 13,mosi,12,clk34,miso,33cs.
Is pin 34 really MISO and pin 12 really clk. I hope you can help.

valera39, 2023-05-17:
Hello [BruceKenley](https://github.com/BruceKenley)
I have a radio on an ESP32 wrower and VS1053. 
MOSI 13 pin, MISO 34 pin and SCK 12 pin works fine, firmware 2.4c

schreibfaul1, 2023-05-17:
If you are using an ESP32 then GPIO 2, 14 and 15 are the pins for SD_MMC and cannot be changed. This fixed assignment does not exist with the ESP32-S3. The GPIOs for SPI can be freely selected with the exception of pins >=34, these are read only and cannot be used for MOSI or CLK. The pin assignment in 'common.h' is just an example, you can change it if necessary.

BruceKenley, 2023-05-22:
Thanks for your replies. I am using a ESP32-vroom-32u. I am still having a problem with VS1053. I am going to try 1k from ground to pin12. I also found if l connected 2nd GND pin of sd card it recognised all types of sd cards. Hope to get working tomorrow.

steve6375, 2023-06-02:
If you are using an SD card holder with no extra circuitry, then it should be connected to **VCC=3.3V** as all SD cards are 3.3V max. and NOT 5V.
If you have an SD arduino shield with a 5V->3.3V regulator then you should not use it because it may also contain 74LVS245 level shifters which means that 5V TTL logic levels will be output to the 3.3V esp32 which is not good!
It is best to use an SD card holder which has no extra circuitry and use 3.3V as VCC to the SD card. Also, I had problems using a Mini-to-Micro card adapter - it was not detected. I had to use a full size SD card (1GB and 4GB both worked as FAT32).
My screen KMRTM32032-SPI comes with a built in mini TF card holder and the connections come straight out to a set of 4 through holes which you can connect the SD signals to (these are not level shifted). So this is ideal - I have not tried this yet on the esp32  however...

closed 2024-02-25 by github-actions[bot]