schreibfaul1/ESP32-MiniWebRadio #167: No sound

valera39, 2023-05-17:
Hello Wolle!
No sound, ESP32S3, VS1053, firmware 2.7.3a. Stations switch but there is no sound and logos are not loaded. Files from the SD card are playable. The touchscreen is very slow if 
SDMMC_FREQUENCY 2000000.
Log: ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x18 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a38
entry 0x403c98d4
E (503) psram: PSRAM ID read error: 0x00ffffff
[   503][W][esp32-hal-psram.c:71] psramInit(): PSRAM init failed!
E (504) esp_core_dump_flash: No core dump partition found!
E (508) esp_core_dump_flash: No core dump partition found!
ESP32 Chip: ESP32-S3
Arduino Version: 2.0.8
ESP-IDF Version: 4.4.4
ARDUINO_LOOP_STACK_SIZE 8192 words (32 bit)

00:00:00
00:00:00        ***************************    
00:00:00        *     MiniWebRadio V2     *    
00:00:00        ***************************    
00:00:00
00:00:00 setup: ....  Arduino is pinned to core 1
00:00:00 tft_info: .  init ILI9488
00:00:00 setup: ....  Init SD card
00:00:00 setup: ....  SD card found
E (315) ledc: ledc_get_duty(739): LEDC is not initialized
00:00:01 setup: ....  seek for stations.csv
00:00:01 setup: ....  stations.csv found
00:00:01 setup: ....  seek for WiFi networks
[  1713][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: ASUS_AC66U
[  1723][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: Honor
[  1725][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: network_01
[  1730][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: network_02
[  1737][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: network_03
[  1745][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: network_04
[  1753][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: network_04
[  8431][I][WiFiMulti.cpp:114] run(): [WIFI] scan done
[  8431][I][WiFiMulti.cpp:119] run(): [WIFI] 7 networks found
[  8432][I][WiFiMulti.cpp:160] run(): [WIFI] Connecting BSSID: 2C:FD:A1:C2:44:F0 SSID: ASUS_AC66U Channel: 12 (-60)
[  8582][I][WiFiMulti.cpp:174] run(): [WIFI] Connecting done.
00:00:08 WiFI_info:   Connecting WiFi...
00:00:08 setup: ....  connected to ASUS_AC66U, IP address is 192.168.1.55
00:00:08 ftpServer:   Buffers allocated: 4622 bytes
00:00:08 rtime_info:  Initializing SNTP
00:00:08 rtime_info:  Waiting for system time to be set... (1/10)
00:00:10 rtime_info:  Waiting for system time to be set... (2/10)
18:52:14 rtime_info:  18:52:14
18:52:14 audiotask:   is pinned to core 1
18:52:14 audiotask:   priority is 2
18:52:14 setup: ....  Number of saved stations: 92
18:52:14 setup: ....  current station number: 9
18:52:14 setup: ....  current volume: 5
18:52:14 setup: ....  last connected host: https://icecast-vgtrk.cdnvideo.ru/vestifm_mp3_192kbps
18:52:14 VS1053 chipID = -1350368918, version = 3
18:52:14 action: ...  current volume is 5
[ 13500][E][Preferences.cpp:483] getString(): nvs_get_str len fail: station_009 NOT_FOUND
18:52:15 action: ...  switch to station 9
18:52:15 AUDIO_info:  Connect to new host: "http://"
18:52:15 AUDIO_info:  Song stopped correctly after 0 msec
18:52:15 AUDIO_info:  PSRAM not found, inputBufferSize: 11903 bytes
[ 14026][E][WiFiGeneric.cpp:1578] hostByName(): DNS Failed for 
18:52:15 AUDIO_info:  Request http:// failed!
18:52:15 StreamTitle:
18:52:16 WiFI_info:   RSSI is -66 dB
18:52:16 DLNA_info:   SSDP search for media servers started
18:52:20 DLNA_info:   SSDP query discovered 1 media servers
18:52:20 DLNA_info:   connected successfully to 192.168.1.1:8200
18:52:20 WiFI_info:   RSSI is -62 dB
18:52:20 ftpServer:   Ftp server waiting for connection on port 21
18:52:27 WiFI_info:   RSSI is -66 dB
[ 27764][E][Preferences.cpp:483] getString(): nvs_get_str len fail: station_010 NOT_FOUND
18:52:29 action: ...  switch to station 10
18:52:29 AUDIO_info:  Connect to new host: "http://"
18:52:29 AUDIO_info:  Song stopped correctly after 0 msec
[ 27804][E][WiFiGeneric.cpp:1578] hostByName(): DNS Failed for 
18:52:29 AUDIO_info:  Request http:// failed!
18:52:29 StreamTitle:
[ 29561][E][Preferences.cpp:483] getString(): nvs_get_str len fail: station_011 NOT_FOUND
18:52:30 action: ...  switch to station 11
18:52:30 AUDIO_info:  Connect to new host: "http://"
18:52:31 AUDIO_info:  Song stopped correctly after 0 msec
[ 29600][E][WiFiGeneric.cpp:1578] hostByName(): DNS Failed for 
18:52:31 AUDIO_info:  Request http:// failed!
18:52:31 StreamTitle:
[ 31296][E][Preferences.cpp:483] getString(): nvs_get_str len fail: station_010 NOT_FOUND
18:52:32 action: ...  switch to station 10
18:52:32 AUDIO_info:  Connect to new host: "http://"
18:52:32 AUDIO_info:  Song stopped correctly after 0 msec
[ 31335][E][WiFiGeneric.cpp:1578] hostByName(): DNS Failed for 
18:52:32 AUDIO_info:  Request http:// failed!
18:52:32 StreamTitle:
18:52:38 Touchpoint not valid x=108, y=187
18:52:38 Touchpoint not valid x=119, y=183
18:52:41 AUDIO_info:  Song stopped correctly after 0 msec
18:53:06 AUDIO_info:  Song stopped correctly after 0 msec
18:53:06 AUDIO_info:  Reading file: "/audiofiles/sample.mp3"
18:53:06 AUDIO_info:  Content-Length: 1954212
18:53:06 AUDIO_info:  ID3 framesSize: 45
18:53:06 AUDIO_info:  ID3 version: 2.4
18:53:06 AUDIO_info:  ID3 normal frames
18:53:06 id3data: ..  SettingsForEncoding: Lavf57.83.100
18:53:06 AUDIO_info:  Audio-Length: 1954167
18:53:06 AUDIO_info:  stream ready
18:53:11 Touchpoint not valid x=286, y=49
[ 72389][E][Preferences.cpp:483] getString(): nvs_get_str len fail: station_010 NOT_FOUND
18:53:14 action: ...  switch to station 10
18:53:14 AUDIO_info:  Connect to new host: "http://"
18:53:14 AUDIO_info:  Song stopped correctly after 20 msec
[ 73064][E][WiFiGeneric.cpp:1578] hostByName(): DNS Failed for 
18:53:14 AUDIO_info:  Request http:// failed!
18:53:14 StreamTitle:
[ 77405][E][Preferences.cpp:483] getString(): nvs_get_str len fail: station_011 NOT_FOUND
18:53:18 action: ...  switch to station 11
18:53:18 AUDIO_info:  Connect to new host: "http://"
18:53:18 AUDIO_info:  Song stopped correctly after 0 msec
[ 77445][E][WiFiGeneric.cpp:1578] hostByName(): DNS Failed for 
18:53:18 AUDIO_info:  Request http:// failed!
18:53:18 StreamTitle:
[ 78947][E][Preferences.cpp:483] getString(): nvs_get_str len fail: station_010 NOT_FOUND
18:53:20 action: ...  switch to station 10
18:53:20 AUDIO_info:  Connect to new host: "http://"
18:53:20 AUDIO_info:  Song stopped correctly after 0 msec
[ 78987][E][WiFiGeneric.cpp:1578] hostByName(): DNS Failed for 
18:53:20 AUDIO_info:  Request http:// failed!
18:53:20 StreamTitle:
Also tried flashing to a blank memory, but all the same
A short video about the problem https://files.fm/u/svrx3mh6f

schreibfaul1, 2023-05-17:
Sorry for the troubles you are having with your board, but ESP32-S3 without PSRAM is rare. Unfortunately, I only found out today that the WiFi client does not want to work with the announced PSRAM. Unfortunately, an entry in the boards/esp32-12k.json prevented the deletion of DBOARD_HAS_PSRAM' from taking effect.

valera39, 2023-05-17:
I am surprised myself, because on the board is clearly visible inscription WROOM-1 N16R8, but on the official Espressif website there is no such a designation, there is WROOM-2 N16R8. And the appearance of the board is different. Most likely it is some kind of copy or fake.
I will try to flash again with the latest changes.

valera39, 2023-05-17:
Genius, you beat it! It worked!
Thank you! I will continue testing

closed 2023-05-19 by valera39