schreibfaul1/ESP32-MiniWebRadio #172: half screen in clock mode

clear-sky, 2023-05-23:
Hi dear Wolle!
in clock mode if you change station  via web interface only half screen is shown - please check!

schreibfaul1, 2023-05-23:
This is the next mistake. I have created a new branch in which this and other errors have already been eliminated. There are also smaller buttons. If it runs stable, this branch will become the new master.
For testing, 'master' can be changed to 'small_buttons' in VSCode at the bottom left of the bar. However, it is necessary to write the new symbols to the SD card, so save the station list and your own logos before unpacking so that they are not lost.

valera39, 2023-05-23:
I'm sorry, maybe this is an inaccurate translation, but I don't understand how to do it. If you can describe it in detail with pictures.
I have this when I press "master"
[image]

closed 2024-02-25 by github-actions[bot]