schreibfaul1/ESP32-MiniWebRadio #174: Is this a bug or not?

valera39, 2023-05-24:
Hello Wolle!
I think I found another bug. In the version with small buttons when playing myPlaylist.m3u from the audiofiles folder, link https://cdn.pixabay.com/audio/2022/01/11/audio_b88f2350c2.mp3 The radio hangs up, the web interface and touchscreen stop working, the port monitor and FTP are unavailable.
After two minutes, the radio comes back to normal. Is it supposed to be like this? 
A little log:
22:22:34 AUDIO_info:  Song stopped correctly after 20 msec
22:22:34 AUDIO_info:  End of file "logo-1.mp3"
22:22:34 end of file: logo-1.mp3
22:22:35 AUDIO_info:  next playlist file
22:22:35 Playlist:    #EXTINF:3,τόνος ανακοίνωσης - ανακοίνωση-3.mp3
22:22:35 Playlist:    Title: τόνος ανακοίνωσης - ανακοίνωση-3.mp3
22:22:35 Playlist:    playtime: 3s
22:22:35 Playlist:    myPlaylist/ανακοίνωση-3.mp3
22:22:35 AUDIO_info:  Song stopped correctly after 0 msec
22:22:35 AUDIO_info:  Reading file: "/audiofiles/myPlaylist/ανακοίνωση-3.mp3"
22:22:35 AUDIO_info:  Content-Length: 156849
22:22:35 AUDIO_info:  ID3 framesSize: 116
22:22:35 AUDIO_info:  ID3 version: 2.4
22:22:35 AUDIO_info:  ID3 extended header
22:22:35 Audiocodec:  MP3
22:22:35 id3data: ..  Album: τόνος ανακοίνωσης
22:22:35 id3data: ..  Artist: τόνος ανακοίνωσης
22:22:35 AUDIO_info:  Audio-Length: 156021
22:22:35 AUDIO_info:  stream ready
22:22:39 AUDIO_info:  Song stopped correctly after 20 msec
22:22:39 AUDIO_info:  End of file "ανακοίνωση-3.mp3"
22:22:39 end of file: ανακοίνωση-3.mp3
22:22:40 AUDIO_info:  next playlist file
22:22:40 Playlist:    #EXTINF:4,фирменный или товарный знак - Лого-2.mp3
22:22:40 Playlist:    Title: фирменный или товарный знак - Лого-2.mp3
22:22:40 Playlist:    playtime: 4s
22:22:40 Playlist:    myPlaylist/Лого-2.mp3
22:22:40 AUDIO_info:  Song stopped correctly after 0 msec
22:22:40 AUDIO_info:  Reading file: "/audiofiles/myPlaylist/Лого-2.mp3"
22:22:40 AUDIO_info:  Content-Length: 163159
22:22:40 AUDIO_info:  ID3 framesSize: 156
22:22:40 AUDIO_info:  ID3 version: 2.4
22:22:40 AUDIO_info:  ID3 extended header
22:22:40 id3data: ..  Album: фирменный или товарный знак
22:22:40 id3data: ..  Artist: фирменный или товарный знак
22:22:40 AUDIO_info:  Audio-Length: 162175
22:22:40 AUDIO_info:  stream ready
22:22:41 Audiocodec:  MP3
22:22:44 AUDIO_info:  Song stopped correctly after 20 msec
22:22:44 AUDIO_info:  End of file "Лого-2.mp3"
22:22:44 end of file: Лого-2.mp3
22:22:45 AUDIO_info:  next playlist file
22:22:45 Playlist:    #EXTINF:8,audio_b88f2350c2.mp3
22:22:45 Playlist:    Title: audio_b88f2350c2.mp3
22:22:46 Playlist:    playtime: 8s
22:22:46 Playlist:    https://cdn.pixabay.com/audio/2022/01/11/audio_b88f2350c2.mp3
22:22:46 AUDIO_info:  Connect to new host: "https://cdn.pixabay.com/audio/2022/01/11/audio_b88f2350c2.mp3"
22:22:46 AUDIO_info:  Song stopped correctly after 0 msec
[847131][E][WiFiClientSecure.cpp:144] connect(): start_ssl_client: -1
22:24:46 AUDIO_info:  Request https://cdn.pixabay.com/audio/2022/01/11/audio_b88f2350c2.mp3 failed!
22:24:46 StreamTitle: 
22:24:51 AUDIO_info:  next playlist file
22:24:51 end of playlist

schreibfaul1, 2023-05-25:
I'm not sure but I suspect that the destination is not achievable. The timeout for SSL is set to 2700ms, see vs1053_ext.h `m_timeout_ms_ssl = 2700` . This is from the old days, the S3 doesn't take that long to establish an SSL connection. Please check whether the page https://cdn.pixabay.com/audio/2022/01/11/audio_b88f2350c2.mp3 can be reached from your location.

valera39, 2023-05-25:
Yeah, it looks like it only works for me through a VPN

closed 2023-06-02 by valera39