schreibfaul1/ESP32-MiniWebRadio #175: Could you please add support 4 inch ST7796S display?

Hhaltt, 2023-05-26:
[Link to display](https://a.aliexpress.com/_Evt0o81)  
Could you please add support for this display? Touch controller is XPT2046.

schreibfaul1, 2023-05-26:
It looks like ST7796S and ILI9486 use the same instruction set. It might work if you select ILI9486b as TFT_CONTROLLER.

Hhaltt, 2023-05-26:
Yes, it works but not correctly.  The best result is if ILI9341 is selected as the controller, but the part of the display is just white.  If I choose ILI9486 - the picture is strongly shifted to the side and about half of the display is also white.  My knowledge does not allow me to fix this

schreibfaul1, 2023-05-26:
If I come across such a display, I will add the ST7796 controller.
It looks like the displays will only ship from China, delivery time 3-4 weeks.
I can't help right now.
If ILI9341 works similarly, you can try changing its height and width, from 320x240 to 480x320. In tft.h line 41 and 42

Hhaltt, 2023-05-26:
OK, you were right.  The display is fully functional: TFT_CONTROLLER : 4 (ILI9488), TP_VERSION : 4 (ILI9488).  It's just that after turning on the power or resetting it does not always work correctly, sometimes it remains just white.  The touchscreen also works.  And if you want to use the SD card slot on the display, you need to solder a pull-up resistor to the sd-miso pin (3.3v), also on my version of the PCM5102A DAC you need to pull-up to 3.3v on the XMT pin

Hhaltt, 2023-05-26:
Thanks, this is a cool project.  And one more little wish - if the stream / track playback time were displayed - it would be very good

schreibfaul1, 2023-06-09:
I added the ST7796 TFT controller to the 'small buttons' branch. This branch will soon become the master. It looks like the display is running stable now.

Hhaltt, 2023-06-23:
I don't know what's wrong with my display. I think the main reason is that he is Chinese. I tried everything possible with ST7796 and it's just white. I went back to ILI9488, it works. Thank you for your work. By the way VU-meter looks great.

closed 2023-06-23 by Hhaltt

schreibfaul1, 2023-06-23:
I tested these two displays, both with ST7796. The darker one is the Raspberry PI version. Despite the same controller, the displays behave completely differently.
[image]

The initialization of the TFT controllers are mostly very similar to ILI9488 (or ILI9486) and differ only in details. And the gamma (color value) correction can be different. But the main thing is that it works!