schreibfaul1/ESP32-MiniWebRadio #176: The last station is not saved

valera39, 2023-06-02:
Hello Wolle!
I found that after switching on, the latest version with small buttons does not save the last station played before.
The first station always plays when you turn it on. The rest of the settings are retained when you turn it off.
The configuration is both ESP32 and ESP32S3. If you select master 08324aee with the big buttons, everything is fine.
Thank you!

valera39, 2023-06-02:
A little clarification, on ESP32-S3 there is no memory of the last played station, on any branch of the latest firmware

schreibfaul1, 2023-06-03:
So far, the settings including lastURL were saved in nvs. I'm in the process of moving it to the SD card. A file named settings.json is created there. Changes are saved to the file every minute. Maybe a minute is too long?

valera39, 2023-06-03:
Thank you for answering, I think a minute is a lot, where can I change the value down?
I think it would be nice if the data were updated immediately after turning on the station.

valera39, 2023-06-17:
Hello Wolle!
Is it possible to reduce the station save time?

closed 2024-02-22 by github-actions[bot]