schreibfaul1/ESP32-MiniWebRadio #177: PSRAM

valera39, 2023-06-02:
Hello Wolle!
I apologize for my many questions, I just want to understand
I use the original board like this[https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/hw-reference/esp32s3/user-guide-devkitc-1.html](url)
[image]

But in the log I get an entry: PSRAM not found, inputBufferSize: 14399 bytes.
Why?

schreibfaul1, 2023-06-03:
That's more of a question for Espressif. I check with psramInit() if a PSRAM is available. If false is returned, the PSRAM has not been initialized.

valera39, 2023-06-03:
Maybe it needs to be initialized in some way, because, as assured by the developers, it should be present.
I would love to use the analogue from Ai Thinker, but unfortunately in my area it is impossible to buy it :(

schreibfaul1, 2023-06-03:
Please run this program:

````c++
#include "Arduino.h"
#include "esp_heap_caps.h"

void setup() {
  Serial.begin(115200);
  delay(1000);

  // Determine the PSRAM size
  size_t psramSize = ESP.getPsramSize();
  Serial.printf("PSRAM size: %d bytes\n", psramSize);

  // Determine the free PSRAM size
  size_t freePsram = heap_caps_get_free_size(MALLOC_CAP_SPIRAM);
  Serial.printf("Free memory in PSRAM: %d bytes\n", freePsram);
}

void loop() {
  ;
}
````

With my ESP12K from Ai-Thinker I get:
[image]

valera39, 2023-06-03:
Checked two ESP32-S3s from Espressif, one of them is original
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a38
entry 0x403c98d4
PSRAM size: 0 bytes
Free memory in PSRAM: 0 bytes

valera39, 2023-06-04:
Maybe it would help you to enable PSRAM in ESP32S3 by Espressif [https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/flash_psram_config.html](url)

valera39, 2023-06-05:
There is no ESP32S3 WROWER, only WROOM, but the ESP32 comes with PSRAM (wrower) and wroom (without PSRAM)

closed 2023-06-09 by valera39